Lora-net/LoRaMac-node #11: recover from LORAMAC_EVENT_INFO_STATUS_ERROR

matijsbrs, 2015-08-18:
Hi,
I'm working on a project using this loraMac-node code. 
In the project I use LoRaMacSendConfirmedFrame in a Class C network on an actility network.
But it seems that the code doesn't recover from a 'LORAMAC_EVENT_INFO_STATUS_ERROR' 
Do you have any suggestions on what to do?

Kind regards,

Matijs

mluis1, 2015-09-07:
Could you please give more details on how to reproduce the issue?

denchii, 2015-11-11:
LoRaMacSendFrameOnChannel function sets error state just after entering, and didn't changes it before exit. Is it has some sense? Cos it's still not changed in 3.4 release.

uint8_t LoRaMacSendFrameOnChannel( ChannelParams_t channel )
{
    **LoRaMacEventInfo.Status = LORAMAC_EVENT_INFO_STATUS_ERROR;**
    LoRaMacEventInfo.TxDatarate = Datarates[ChannelsDatarate];
.....
}

mluis1, 2015-11-11:
@denchii 
The variable value is changed when on of the callbacks functions (OnRadioTxDone, OnRadioRxDone,  etc..) are called.

The variable must be initialized and currently the LoRaMacSendFrameOnChannel looks to be the right place to do it.

@matijsbrs 
Have you tried with the latest version? It should solve the issue.

denchii, 2015-11-11:
@mluis1 
Got it, thanks.

closed 2015-12-18 by mluis1