Lora-net/LoRaMac-node #12: Relation to IBM's LMIC

matthijskooijman, 2015-08-22:
Quick question: How does this implementation relate to IBM's LMIC library? Version 3.1 of this library stated that the next version of this library would be based on IBM's LMIC library, but I'm not sure if that change actually happened (a quick look at the code shows the code structure is different from LMIC still). Is it now based on LMIC? Or was the plan to switch over cancelled?

pdemil, 2015-08-23:
Semtech stack is not based on IBM's LMIC library. The plan to switch is cancelled: see remark 1 at https://github.com/Lora-net/LoRaMac-node/tree/v3.2 :" REMARK 1: It has been decided to continue the maintenance of the Semtech LoRaWAN implementation."

matthijskooijman, 2015-08-23:
@pdemil, thanks for confirming, I missed that remark looking through the history.

Also, when we spoke last friday, this is exactly why I didn't remember there was a second implementation - when I looked at it, the intention was to switch over, so I disregarded it at that time. Having seen this now, I might have a go at porting this library to Arduino instead of LMIC.

In any case, my question is answered, closing.

closed 2015-08-23 by matthijskooijman