Lora-net/LoRaMac-node #16: US 902-928MHz ISM Band

FoujanetG, 2015-10-07:
Hello,

I would like to know if the stack is working with the US902-928MHz ISM Band.
There is some code with "USE_BAND_915". The stack is it fully compatible with the US 915MHz band?
What are the limitations between the implementation and the specification LoraWan1.0 ?
 All DR are not implemented for example. Just DR with BW125 exist. 

Do you have a plan to include the  915MHz band in a next version?

Thanks a lot.
GuillaumeF

mluis1, 2015-10-07:
Hello,

The US915 band isn't yet implemented on the current version.
A version fully supporting the US915 band is currently under test.
The next release main goal is to add the US915 support.

Best regards,

Miguel

FoujanetG, 2015-10-07:
Many thanks for the prompt reply.
Do you have an idea for the next release date ?

Best regards,
Guillaume F.

mluis1, 2015-10-07:
We hope to do it somewhere in the coming couple of weeks.

Best regards,
Miguel

mleksio, 2015-10-24:
Hi! I have two questions... have you tried to connect node with SW based on your LoRaWan implmentation with Multitech Conduit on 915MHz?
Is there chance to get beta version of this stack with 915MHz support?

closed 2015-10-30 by mluis1