Lora-net/LoRaMac-node #26: Class C - Receive on first window

reissjason, 2015-11-10:
A Class C device will open a window with rx2 parameters in the TxDone function.
When Rx1Window timeout occurs will the rx1 window be able to open when radio not idle?

mluis1, 2015-11-11:
We think that you may be correct. We will check this in the next few days.

rossihwang, 2015-11-12:
I found this problem too and others, I'm trying fix these in my fork

AlexanderWiniger, 2015-11-12:
A quick fix would be to add 

if (LoRaMacDeviceClass == CLASS_C) {
        Radio.Standby();
}

at the beginning of the function OnRxWindow1TimerEvent. But there is another problem that RX2 will not be opened correctly. There seem to be several problem with the class C implementation

mluis1, 2015-11-12:
Currently this is the fix that I have also implemented. I'm currently testing the Class C implementation. When I have tested everything I'll make a new release.

AlexanderWiniger, 2015-11-12:
Updated previous comment with additional fixes

mluis1, 2015-11-12:
@AlexanderWiniger 
I don't agree with the following statement:
"Then it ought to be re-opened after RX2_DELAY and stay open until you want to send the next time. At the moment in class C mode the RX2 window timer is not activated. In order to fix this problem one has to remove the if-clause in the OnRadioTxDone function"

What we intended when the Class C has been specified is that when you are not transmitting or handling the 1st reception window the end-device is setup to handle the 2nd reception window. The current implementation does exactly this.
Once the `OnRadioTxDone` function is called we call `OnRxWindow2TimerEvent( );` which setups the reception window with Rx2 parameters. 
```C
static void OnRadioTxDone( void )
{
    TimerTime_t curTime = TimerGetCurrentTime( );
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }
```
When the `OnRxWindow1TimerEvent( );` is called the the Rx1 reception parameters are set. They will last up until we have an `OnRxDone` or `OnRxTimeout` or `OnRxError`. After those events the radio must be setup with Rx2 parameters.

The implementation that I'm currently testing the `OnRxTimeout` and `OnRxError` functions are as follows: (Please note that the else case of device class check has been added)
```C
/*!
 * Function executed on Radio Rx Timeout event
 */
static void OnRadioRxTimeout( void )
{
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }
    if( LoRaMacEventFlags.Bits.RxSlot == 1 )
    {
        LoRaMacEventFlags.Bits.Tx = 1;
        LoRaMacEventInfo.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;
    }
}

/*!
 * Function executed on Radio Rx Error event
 */
static void OnRadioRxError( void )
{
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }
    if( LoRaMacEventFlags.Bits.RxSlot == 1 )
    {
        LoRaMacEventFlags.Bits.Tx = 1;
        LoRaMacEventInfo.Status = LORAMAC_EVENT_INFO_STATUS_RX2_ERROR;
    }
}
```

JiapengLi, 2015-11-12:
Should the extra RXWIN2 between RXWIN1 and TXDONE be able to receive packets which are longer than RX1Delay(eg: 1s)?

closed 2015-12-18 by mluis1