Lora-net/LoRaMac-node #29: SNR value range

reissjason, 2015-11-19:
From the OnRadioRxDone handler I see values from -57 to 34. Is this value in cB or dB? 

I see the SxRadio has divided the register value by 4 and this looks correct from the Sx1272 documentation. But these values maybe too high for dB and too low for cB.

Also in OnRadioRxDone the SNR value is then put into the event object which has unsigned integer type.

mluis1, 2015-11-25:
The SNR value returned by the OnRxRadioDone callback function is the radio register raw value.
The reason is that the LoRaWAN specification requests to have the register raw value.

So, when using the SNR value returned by the callback function you must apply the division by 4 as on the radio driver in order to get the real SNR value.

The radio driver implements the division by 4 in order to compute the RSSI value when we are below the noise level.

closed 2015-12-18 by mluis1

Mobiak, 2016-04-26:
In file sx1272.c when device receive a message, it use snr value only to calculate the RSSI. The value transmitted by callback to LoRaMac is the raw value directly read from the SX1272 register.
But in the stack LoRaWan, there is no transformation of this value, it is stored and given back to the network layer as is.
The issue is that the LoRaWan specification R1.0  and the LoRaWAN1.0.1final_876_3 are both saying:

_"The margin (Margin) is the demodulation signal-to-noise ratio in dB rounded to the nearest 8 integer value for the last successfully received DevStatusReq command. It is a signed 9 integer of 6 bits with a minimum value of -32 and a maximum value of 31."_

So, should the LoRaWan layer implement the transformation to get the refined value on 6 bit in order to enforce LoRaWan specification ?

sx1272.c - function SX1272OnDio0Irq( ) : reading the value and transformed it for rssi calculation
```

        SX1272.Settings.LoRaPacketHandler.SnrValue = SX1272Read( REG_LR_PKTSNRVALUE );
        if( SX1272.Settings.LoRaPacketHandler.SnrValue & 0x80 ) // The SNR sign bit is 1
        {
            // Invert and divide by 4
            snr = ( ( ~SX1272.Settings.LoRaPacketHandler.SnrValue + 1 ) & 0xFF ) >> 2;
            snr = -snr;
        }
        else
        {
            // Divide by 4
            snr = ( SX1272.Settings.LoRaPacketHandler.SnrValue & 0xFF ) >> 2;
        }

        int16_t rssi = SX1272Read( REG_LR_PKTRSSIVALUE );
        if( snr < 0 )
        {
            SX1272.Settings.LoRaPacketHandler.RssiValue = RSSI_OFFSET + rssi + ( rssi >> 4 ) +
                                                          snr;
        }
        else
        {    
            SX1272.Settings.LoRaPacketHandler.RssiValue = RSSI_OFFSET + rssi + ( rssi >> 4 );
        }

        SX1272.Settings.LoRaPacketHandler.Size = SX1272Read( REG_LR_RXNBBYTES );
        SX1272ReadFifo( RxBuffer, SX1272.Settings.LoRaPacketHandler.Size );
    
        if( SX1272.Settings.LoRa.RxContinuous == false )
        {
            SX1272.Settings.State = RF_IDLE;
        }
        TimerStop( &RxTimeoutTimer );

        if( ( RadioEvents != NULL ) && ( RadioEvents->RxDone != NULL ) )
        {
            RadioEvents->RxDone( RxBuffer, SX1272.Settings.LoRaPacketHandler.Size, SX1272.Settings.LoRaPacketHandler.RssiValue, SX1272.Settings.LoRaPacketHandler.SnrValue );
        }

```

LoRaMac.c - function OnRadioRxDone( ) : prototype

`        static void OnRadioRxDone( uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr )`

LoRaMac.c - function OnRadioRxDone( ) : store the value

`        LoRaMacEventInfo.RxSnr = snr;`

LoRaMac.c - function LoRaMacProcessMacCommands( ) : use the previously stored value
```

        case SRV_MAC_DEV_STATUS_REQ:
            {
                uint8_t batteryLevel = BAT_LEVEL_NO_MEASURE;
                if( ( LoRaMacCallbacks != NULL ) && ( LoRaMacCallbacks->GetBatteryLevel != NULL ) )
                {
                    batteryLevel = LoRaMacCallbacks->GetBatteryLevel( );
                }
                AddMacCommand( MOTE_MAC_DEV_STATUS_ANS, batteryLevel, LoRaMacEventInfo.RxSnr );
            }
            break;
```