Lora-net/LoRaMac-node #31: Why EUi are in Little Endian and Keys in Big Endian

FoujanetG, 2015-12-04:
Hi,

I would like to know (for my knowledge), why DevEUi and AppEUI are in Little Endian and keys are in BigEndian?

What is the technical reason?

Thanks

GF.

mluis1, 2015-12-07:
The reason to have little endian is that most processors on market are little endian. So, implementing the arrays and values as little endian numbers makes the compiled code smaller as the data is already organized in the memory in a natural way for the processor.

The drawback is that little endian is not the natural way for Humans.

The LoRaWAN specification states that multi bytes inside the frames are to be sent **over the air** as little endian.
So, what could be implemented is to change the MAC layer APIs in order for the upper layers to use Big endian and then it is up to the API function to reverse the bytes order.

The keys are endianness agnostic, these should be true 128 bits random numbers.

mluis1, 2015-12-18:
In V4.0.0 version EUIs are reversed by the MAC layer.

closed 2015-12-18 by mluis1