Lora-net/LoRaMac-node #32: FSK NewChannel with ACK and ADR

reissjason, 2015-12-15:
If ADR is enabled and a new channel is created by the network server on the device to use FSK.
Then the device's SNR is such that the datarate is raised to FSK mode.
When ACKRetries's are set to 1 or 2 and ACKRetries are exhausted the device will ReEnableDefaultChannels and TxDatarate will remain at FSK mode without a channel enabled.
This will cause the device to be unable to transmit or change datarate without intervention.

It is also an issue if ReEnableDefaultChannels is called after the FSK channel was created when TxDatarate is below FSK mode. When it again reaches the FSK threshold the LinkADRReq will be rejected because the FSK channel is disabled.

djaeckle, 2015-12-16:
Hi reissjason,

> If ADR is enabled and a new channel is created by the network server on the device to use FSK.
> Then the device's SNR is such that the datarate is raised to FSK mode.
> When ACKRetries's are set to 1 [...]

I was not able to reproduce the issue yet. Could you please provide some more information on how to reproduce it?

reissjason, 2015-12-16:
I see it is my error in reading. 
This line will **not** disable the other channels, it will ensure that the first three are enabled.
ChannelsMask[0] = ChannelsMask[0] | ( LC( 1 ) + LC( 2 ) + LC( 3 ) );

I was looking at earlier code that looked like this:
ChannelsMask |= ( LC( 1 ) + LC( 2 ) + LC( 3 ) );

closed 2015-12-16 by reissjason