Lora-net/LoRaMac-node #37: number of retries with sendConfirmed frame

FoujanetG, 2016-01-15:
Hi,

I think there is a problem with the dafault value of AckTimeoutRetriesCounter.

When I configure LoRaMacSendConfirmedFrame with nbRetries=1,  the frame is sent one time( with ACK timeOut of course). and if configure  nbRetries=2, the frame is sent two times.

Normally, the number of retrie is in more of the first frame. So if the value of nbRetries is 3, 4 frames are sent if there is no Acknowledge.

Thanks

GF.

jmalangoni, 2016-01-15:
iirc if nbRetries = 0 the value reverts to some default. so nbRetries actually represents "number of retries - 1"? If so, maybe this just needs to be documented better...

djaeckle, 2016-01-22:
Hi,

thanks for the report. Please intrepret the attribute "nbRetries" as the "number of trials". We will update the name of the attribute "nbRetries" and also the related documentation in the next release.

closed 2016-03-11 by mluis1