Lora-net/LoRaMac-node #38: SX1272GetPaSelect is not board specific

timothyjohncarney, 2016-01-19:
Why does SX1272GetPaSelect always return RF_PACONFIG_PASELECT_PABOOST?

According to the comment, RF_PACONFIG_PASELECT_RFO should be the default. 

Furthermore, I believe this should be a variable that is set in the board-specific board.h file. I am working on a design that does not use the PA_BOOST pin so I must change sx1272-board.c in order to have the correct value written to register 9 on transmit. Instead, can SX1272GetPaSelect check to see if a board-specific value is set before setting to a default value?

Thanks,
Tim

mluis1, 2016-02-22:
The return value of SX1272GetPaSelect is dependent on the board you use. For example:
- The LoRaMote board returns always RF_PACONFIG_PASELECT_PABOOST because only PaBosst pin is connected to the antenna.
```C
uint8_t SX1272GetPaSelect( uint32_t channel )
{
    return RF_PACONFIG_PASELECT_PABOOST;
}
```
- The SensorNode board will return either PaBoost or RFO depending on the frequency band used.
```C
uint8_t SX1276GetPaSelect( uint32_t channel )
{
    if( channel < RF_MID_BAND_THRESH )
    {
        return RF_PACONFIG_PASELECT_PABOOST;
    }
    else
    {
        return RF_PACONFIG_PASELECT_RFO;
    }
}
```
So the SX127xGetPaSelect must be modified according to which pin you have connected to the antenna on your board design.

The "x-board.x" files are meant to be customized according to your specific board requirements.

In order to adapt this project to your specific board you should:
1. Create a directory with your board name under "src\board" directory
2. Copy the files from another board as an example. The minimum set of files to be copied are board.c/h, gpio-board.c/h, timer-board.c/h, spi-board.c/h and sx127x-board.c/h. Then if you make usage of other functionality such the I2C then you must also implement an i2c-board.c/h for instance.
3. Implement the contents of each function on each file according to your specific board. Take care to not modify the functions prototypes.
4. Don't forget also to create a specific project to your board.

closed 2016-03-11 by mluis1

Mehradzie, 2017-08-15:
@mluis1 , would you care to elaborate why on earth the code decides to turn the PABOOST on if the channel is lower than 525Mhz?

The SensorNode board will return either PaBoost or RFO depending on the frequency band used.

> uint8_t SX1276GetPaSelect( uint32_t channel )
> {
>     if( channel < RF_MID_BAND_THRESH )
>     {
>         return RF_PACONFIG_PASELECT_PABOOST;
>     }
>     else
>     {
>         return RF_PACONFIG_PASELECT_RFO;
>     }
> }

I can't find which part of Semtech datasheet talks about this decision. 
From what I can find in the SX1276 datasheet you have a choice to turn the PABOOST on or off based on power consumption levels etc.

That would be much appreciated.

rafacouto, 2019-08-23:
@Mehradzie According to datasheet 5.4, power amplifiers: _PA\_BOOST pin_ is used only in low band transmission.

[image]