Lora-net/LoRaMac-node #40: Debug Function

rayaan92, 2016-01-25:
Hi, 

I need some assistance with setup the usart for serial debugging, I am recently worked on the LMIC libraries from IBM but was told that is official outdated and that I should try to port my code over to the semtech LoRaMac instead, any help will be highly appreciated. Thank you in advance.

djaeckle, 2016-01-26:
Hi rayaan92,

the repository contains drivers for the usart. You can also find board specific hardware drivers and setup examples.

rayaan92, 2016-01-27:
thanks, I have had a look at it and have given it a try but I can't seem to get it up and running and I need it urgently. I am working on the SK-IM880 platform, and once I try and configure the uart the code just hangs. here's a snippet of the debug code that i have written.

void debugmcu( void)
	
{
	      UartInit( &Uart1, UART_1, UART_TX, UART_RX );
        UartConfig( &Uart1, RX_ONLY, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );
}

void debugInit ( void )
{
	
	FifoInit( &Uart1.FifoTx, TxBuffer, FIFO_TX_SIZE );
	
}

void debug_char(uint8_t c)
{
	UartMcuPutChar( &Uart1, c);
	
}

void debug_str(const uint8_t* str)
{
	while(*str)
	{
		debug_char(*str++);
	}

mluis1, 2016-02-22:
One of the issues that I see in your code example is that you configure the UART as RX_ONLY but you are trying to send data. So, you should change the `debugmcu` function as follows
```C
void debugmcu( void)
{
    UartInit( &Uart1, UART_1, UART_TX, UART_RX );
    UartConfig( &Uart1, TX_ONLY, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );
}
```
Most of the UART functions return status values. Those return values should be checked in order to verify that the UART peripheral is not busy performing some other operation.
In your code you initialize the transmission FIFO but, to print debug messages most of the time there is no need to do it. Using a FIFO will consume some amount of RAM memory that may be necessary for your application.

Please note also that printing debug messages on interrupt handling functions is not recommended. Print functions are quite time consuming while an interrupt handling function must be handled as quick as possible.

Please find below a simple example to perform an UART echo. I didn't test the example nor compiled it but I guess that it is correct.
```C
int8_t welcomeStr = "Echo application. Every typed char is echoed!!!\r\n";

int main ( void )
{
    uint8_t data = 0;

    UartInit( &Uart1, UART_1, UART_TX, UART_RX );
    UartConfig( &Uart1, RX_TX, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );

    // Print welcome message
    while( UartPutBuffer( &Uart1, welcomeStr, strlen( welcomeStr ) ) != 0 ){ };

    while( 1 )
    {
        if( UartGetChar( &Uart1, &data ) == 0 )
        {
             while( UartPutChar( &Uart1, data) != 0 ){ };
        }
    }
}
```

closed 2016-03-11 by mluis1

rayaan92, 2016-03-22:
Hi Miguel,

I built the snippet code you gave me and it does npt want to build, however
i am using it as a guideline, I am currently working on the sk-im880a class
A LoRamac from you, however when I try to the uart interface the board goes
into a weird state and it seems like it just halts at that point. belong is
the snippet of my code, I do the initialization in the BoardInitMcu
function.

void BoardInitMcu( void )
{
    if( McuInitialized == false )
    {
        // We use IRQ priority group 4 for the entire project
        // When setting the IRQ, only the preemption priority is used
        NVIC_PriorityGroupConfig( NVIC_PriorityGroup_4 );

        // Disable Systick
        SysTick->CTRL  &= ~SysTick_CTRL_TICKINT_Msk;    // Systick IRQ off
        SCB->ICSR |= SCB_ICSR_PENDSTCLR_Msk;            // Clear SysTick
Exception pending flag

        AdcInit( &Adc, POTI );

        UartInit( &Uart1, UART_1, UART_TX, UART_RX);
        UartConfig(&Uart1, RX_TX, 115200, UART_8_BIT, UART_1_STOP_BIT,
NO_PARITY, NO_FLOW_CTRL); (once I comment this line of code out the rest of
the code executes)

        SpiInit( &SX1272.Spi, RADIO_MOSI, RADIO_MISO, RADIO_SCLK, NC );
        SX1272IoInit( );

#if( LOW_POWER_MODE_ENABLE )
        TimerSetLowPowerEnable( true );