Lora-net/LoRaMac-node #42: no getter for MIB_UPLINK_COUNTER, MIB_REPEATER_SUPPORT missing from doc

jmalangoni, 2016-01-25:
getter for MIB_UPLINK_COUNTER isn't implemented in LoRaMacMibGetRequestConfirm(), although should be included based on comment block from LoRaMac.h reproduced below.

Also noticed MIB_REPEATER_SUPPORT is missing entirely from comment block below, although getter and setter are both implemented.
```c
/*!
 * LoRa Mac Information Base (MIB)
 *
 * The following table lists the MIB parameters and the related attributes:
 *
 * Attribute                         | Get | Set
 * --------------------------------- | :-: | :-:
 * \ref MIB_DEVICE_CLASS             | YES | YES
 * \ref MIB_NETWORK_JOINED           | YES | YES
 * \ref MIB_ADR                      | YES | YES
 * \ref MIB_NET_ID                   | YES | YES
 * \ref MIB_DEV_ADDR                 | YES | YES
 * \ref MIB_NWK_SKEY                 | YES | YES
 * \ref MIB_APP_SKEY                 | YES | YES
 * \ref MIB_PUBLIC_NETWORK           | YES | YES
 * \ref MIB_CHANNELS                 | YES | NO
 * \ref MIB_RX2_CHANNEL              | YES | YES
 * \ref MIB_CHANNELS_MASK            | YES | YES
 * \ref MIB_CHANNELS_NB_REP          | YES | YES
 * \ref MIB_MAX_RX_WINDOW_DURATION   | YES | YES
 * \ref MIB_RECEIVE_DELAY_1          | YES | YES
 * \ref MIB_RECEIVE_DELAY_2          | YES | YES
 * \ref MIB_JOIN_ACCEPT_DELAY_1      | YES | YES
 * \ref MIB_JOIN_ACCEPT_DELAY_2      | YES | YES
 * \ref MIB_CHANNELS_DATARATE        | YES | YES
 * \ref MIB_CHANNELS_TX_POWER        | YES | YES
 * \ref MIB_UPLINK_COUNTER           | YES | NO
 * \ref MIB_DOWNLINK_COUNTER         | YES | NO
 * \ref MIB_MULTICAST_CHANNEL        | YES | NO
 *
 * The following table provides links to the function implementations of the
 * related MIB primitives:
 *
 * Primitive        | Function
 * ---------------- | :---------------------:
 * MIB-Set          | \ref LoRaMacMibSetRequestConfirm
 * MIB-Get          | \ref LoRaMacMibGetRequestConfirm
 */
```

djaeckle, 2016-01-25:
Hi jmalangoni,

thanks for the report. Yes, MIB_REPEATER_SUPPORT is missing in the function documentation. We will update it with the next release. I double checked the getter for MIB_UPLINK_COUNTER and it is implemented actually.

jmalangoni, 2016-01-25:
apologies, I must have mistaken the missing setter for a missing getter. thanks

closed 2016-01-25 by jmalangoni