Lora-net/LoRaMac-node #43: New channel Request error

FoujanetG, 2016-01-27:
Hi,

I use the new stack, with the wrapper for api_V3.

The bug is in class A and class C.

When the network send a new channel(MAC cmd ```SRV_MAC_NEW_CHANNEL_REQ```),
all new channels are refused. 
the function LoRaMacChannelAdd() returns ``` LORAMAC_STATUS_BUSY.```

I think, the bug can be fixed if you add  the line``` LoRaMacState &= ~MAC_TX_RUNNING;``` 
 in LoRaMac.c, /```OnRadioRxDone()``` / ```case FRAME_TYPE_DATA_UNCONFIRMED_DOWN:``` / ```  if( isMicOk == true )``` / ```else``` of ```if( multicast == 1 )" / ``` out of ``` if( ( DownLinkCounter == downLinkCounter ) &&
                            ( DownLinkCounter != 0 ) )```

The second pb with this bug, is the stack uses theses new channels, despite the error. 

Thanks

GF.

Smarco10, 2016-01-27:
Hi,

I also have encountered this issue, today, and conclude to the same things.
Thanks in advance for the fix.

Marc-Antoine MARTIN

djaeckle, 2016-01-27:
Hi FoujanetG,

thanks for the report. I was able to reproduce the issue. We will fix the issue with the upcoming release.

closed 2016-03-11 by mluis1