Lora-net/LoRaMac-node #48: CFList issue with FRAME_TYPE_JOIN_ACCEPT

Mobiak, 2016-02-25:
Device fail to add the CFList transmitted in a join_accept frame.

When joining on OTA and getting a join_accept, the device will execute function LoRaMacChannelAdd( i, param ) to add incoming channels. This function start with few checks and fail to succeed this one :

    if( ( LoRaMacState & MAC_TX_RUNNING ) == MAC_TX_RUNNING )
    {
        return LORAMAC_STATUS_BUSY;
    }

The LoRaMacState is MAC_TX_RUNNING because it's sending join_request and getting an join_accept.
The flag can not be cleared before as it is running. Is it possible to modify this check?

mluis1, 2016-02-25:
Duplicate of Issue #43

Mobiak, 2016-02-25:
My bad on this one, forgot the previous posts and didn't check.

closed 2016-03-11 by mluis1