Lora-net/LoRaMac-node #52: MaxPayloadOfDatarate[] wrong order

ruanolin, 2016-03-08:
The vector is defined as:
const uint8_t MaxPayloadOfDatarate[] = { 51, 51, 51, 115, 242, 242, 242, 242 };

But in prepareframe function we put:
ValidatePayloadLength( fBufferSize, ChannelsDatarate, MacCommandsBufferIndex )

In the case ChannelsDatarate = DR0 => SF12, ValidatePayloadLength function will compare the buffer size with 51 when must be compared with 242.

Is this correct or I'm wrong?

pdemil, 2016-03-09:
Maximum application payload on channel datarate DR0 is 51 (not 242), so the ValidatePayloadLength function is correct.

ruanolin, 2016-03-09:
But DR0 is equal to spreading factor 12, no?
So according to the specifications the maximum payload size is 242.

Thanks and sorry if I'm wrong.

pdemil, 2016-03-09:
Yes, DR0 is using SF12.
https://www.lora-alliance.org/portals/0/specs/LoRaWAN%20Specification%201R0.pdf
Have a look at Table 17: EU863-870 maximum payload size (not repeater compatible), 
DR0: N is 51 

Cheers,
Pieter

ruanolin, 2016-03-09:
It's true! I read the specifications a lot of times, but I got confused with the DataRate and Spreading factors.
Thanks!

closed 2016-03-11 by mluis1