Lora-net/LoRaMac-node #53: Bootloader Issue

Smarco10, 2016-03-11:
Hi all,

Trying to program the LoRaMote through DFU, I faced a problem with the bootloader project.
At the compilation LD time, I got an error telling me that the .text section overflow the memory mapped  size.

Checking into the ".ld" [https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/LoRaMote/cmsis/arm-gcc/stm32l1xx_md_flash_bootloader.ld](file), I found an issue at line 4:
`/* Memory regions.*/ 
MEMORY
{
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 12K
  RAM (rwx)  : ORIGIN = 0x20000000, LENGTH = 16K
}`

The Flash length is only 12K instead of 128K. It results in a compilation faillure with  more than 5K overflow.

Thank you in advance for the fix.

Rgds.
MarcAntoine MARTIN

mluis1, 2016-03-14:
A bootloader is small piece of software that provides a way to either update the main application or to directly start the one that is already programmed. The 12K value means that the bootloader application must be smaller than this value.

Please keep in mind that the bootloader and the normal application must live together on the FLASH memory.

Thus, the solutions that you have are the following:
1. Ensure that the bootloader application is smaller than 12K. Please verify compiler optimization, if you have added code to the project try to reduce the occupied size of your adding.
2. In your case it looks like you need 5K more. You may change the LD value to 17K but then don't forget to update the vectors in the main application LD file as well as the interrupt vector.
```C
void BoardInitMcu( void )
{
    if( McuInitialized == false )
    {
#if defined( USE_BOOTLOADER )
        // Set the Vector Table base location at 0x3000
        NVIC_SetVectorTable( NVIC_VectTab_FLASH, 0x3000 );
#endif
        // We use IRQ priority group 4 for the entire project
        // When setting the IRQ, only the preemption priority is used
        NVIC_PriorityGroupConfig( NVIC_PriorityGroup_4 );

...
```

The bigger is the bootloader application the less space you will have for your normal application.

Smarco10, 2016-03-14:
Thank you mluis1 for your reply.

I didn't change the code, I compiled it with the CoIDE project provided by Semtech and I still have the 5K-more-required error.

I changed the value to 20K, it well compiled but when I plug the LoRaMote on USB, I get a DFU driver error (I installed the STMicroelectronic DFU Driver furnished with DFuse application as mentioned in this document: [http://www.semtech.com/images/datasheet/User_Guide_LoRaMote_STD.pdf](User_Guide_LoRaMote_STD.pdf) (section 5.1.2 - update though USB))

Actually I get the Windows Driver error 10: can't start the device. LEDs blink only once (if we look at the main.c file, there is an infinit loop which should make the LED blinking infinitly).

Do you have any information about this issue?

Rgds

mluis1, 2016-03-14:
I have just compiled the code using coIDE and download the hex file to a LoRaMote without problems. I'm using version 2.0.1 of coIDE.
At the end of the compilation I have the following output:

> Program Size:
>       text	   data	    bss	    dec	    hex	filename
>      10764	   1136	   1340	  13240	   33b8	BootLoader.elf
> 
> BUILD SUCCESSFUL
> Total time: 12 seconds
> 

Except maybe differences on the version of the coIDE application I don't have any more information on the issue.

Smarco10, 2016-03-16:
Hi Miguel,

I use the version 2.0.3 of CoIDE so I don't think the problem come from the version.
Hereafter is the result of compilation:
`[cc] arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -g2 -Wall -Os -std=c99 -c -DUSE_NO_TIMER [...]
       [cc] C:\Users\MARC-A~1\AppData\Local\Temp\ccuF66uS.s: Assembler messages:
       [cc] C:\Users\MARC-A~1\AppData\Local\Temp\ccuF66uS.s:18: Warning: ignoring changed section attributes for .data
       [cc] 0 total files to be compiled.
       [cc] Starting link
       [cc] arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -g2 -nostartfiles -Wl,-Map=BootLoader.map -Os -Wl,--gc-sections --specs=nano.specs [...] -g -o BootLoader.elf -lm
Program Size:
      text	   data	    bss	    dec	    hex	filename
     17500	   1152	   1336	  19988	   4e14	BootLoader.elf

BUILD SUCCESSFUL
Total time: 12 seconds`

I don't understand why all my sections are bigger than yours. Espacially the text section which is almost twice bigger. I tried to change optimisation settings but it already size-optimized.
The stack version I use is 4.1.0 and the arm crosscompile toolchain version is: 

> gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (GNU Tools for ARM Embedded Processors)

I'm highly interested to understand where am I wrong.
Thank you for the time you spend to assist me.

Rgds

mluis1, 2016-03-16:
Currently I'm using the following version for the GCC compiler

> arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.8.4 20140526 (release) [ARM/embedded-4_8-branch revision 211358]

Maybe the math library or the C library is bigger in your more recent version.

Smarco10, 2016-03-16:
Size appart,
I succeed to compile it (20KB instead of 12KB) and upload it into the LoRaMote but what is wierd is that I does not want to mount on Windows, I get the driver error 10.

I'll continue to keep flashing with STLink but that's wierd.

Which DFU driver do you use on Windows?

mluis1, 2016-03-16:
The latest version furnished by ST at the following link [DfuSe USB](http://www.st.com/web/en/catalog/tools/FM147/CL1794/SC961/SS1533/PF257916#)

closed 2016-04-15 by mluis1