Lora-net/LoRaMac-node #54: PingPong example in FSK mode

FoujanetG, 2016-03-14:
Hi,

I am trying the FSK mode with the Ping Pong Example.
With a buffer size = 64, it's working well.

Then I try with a buffer longer(size = 200).( All bytes are filled)
The radio transfer works, 32 first bytes are ok, but bytes 33 to 199 are egual to zero.

Are you already checked the FSK mode with the radio driver and a frame longer than 64 bytes?

Thanks,
Guillaume.

thomatec, 2016-03-15:
Hi, 

FSK Fifo is 64 byte length. With fifolevel interrupt, SX1272 driver can begin read fifo before it is full using DIO1. This allows transmit frame over 64 bit.

Check  REG_FIFOTHRESH  value in your init and check if your are using DIO1.

Thomas

FoujanetG, 2016-03-15:
Hi,

thanks for you response.
Yes I use the new release, and DIO1 is configure as DIO0. 
But the register REG_FIFOTHRESH is apparently not used for the configuration in the sx1276.c file.
Have you tried  the ping pong example in FSK Mode and a bugger longer than 64 bytes?

thanks a lot,
Guillaume

mluis1, 2016-03-15:
Actually we have forgot to add the following line to the SX1276 driver. The SX1272 driver has it.
```C
memcpy1( RxTxBuffer, buffer, size );
```

The function to be modified is the follwoing
```C
void SX1276Send( uint8_t *buffer, uint8_t size )
{
    uint32_t txTimeout = 0;

    switch( SX1276.Settings.Modem )
    {
    case MODEM_FSK:
        {
            SX1276.Settings.FskPacketHandler.NbBytes = 0;
            SX1276.Settings.FskPacketHandler.Size = size;

            if( SX1276.Settings.Fsk.FixLen == false )
            {
                SX1276WriteFifo( ( uint8_t* )&size, 1 );
            }
            else
            {
                SX1276Write( REG_PAYLOADLENGTH, size );
            }            
            
            if( ( size > 0 ) && ( size <= 64 ) )
            {
                SX1276.Settings.FskPacketHandler.ChunkSize = size;
            }
            else
            {
                memcpy1( RxTxBuffer, buffer, size );
                SX1276.Settings.FskPacketHandler.ChunkSize = 32;
            }

            // Write payload buffer
            SX1276WriteFifo( buffer, SX1276.Settings.FskPacketHandler.ChunkSize );
            SX1276.Settings.FskPacketHandler.NbBytes += SX1276.Settings.FskPacketHandler.ChunkSize;
            txTimeout = SX1276.Settings.Fsk.TxTimeout;
        }
        break;
```

We will apply this patch to the develop branch as soon as possible.

FoujanetG, 2016-03-15:
Thank you,
it's working.

Guillaume.

FoujanetG, 2016-03-15:
the patch works.
"We will apply this patch to the develop branch as soon as possible."

FoujanetG, 2016-03-15:
sorry, I re-open the comment.

The patch work with a size buffer = 120;
But it doesn't work  with a buffer size = 250.

What is the maximum buffer size I can use?
I thought that it was 256.

thanks,
guillaume

mluis1, 2016-09-23:
Can we close this issue?

closed 2017-01-24 by mluis1

mwruschel, 2018-11-23:
how to use sx1278 in gmsk mode?

I need to receive data from a radio ( http://pacificcrest.com/library/DS_ADL_Vantage35.pdf ) with the Sx1278 using teensy.

nevercast, 2018-12-17:
@mwruschel I would suggest opening a new issue instead of commenting on this closed one.