Lora-net/LoRaMac-node #55: ADR command

coljander, 2016-03-15:
I've noticed that the ChannelsMask[0] value might not be handled correctly in EU mode in the ProcessMacCommands() routine.  It is originally set in the following line
                        channelsMask[0] = chMask;
but later in the routine there is a section of code that I'm getting into 
                    if( ( status & 0x07 ) == 0x07 )
                    {
                        ChannelsDatarate = datarate;
                        ChannelsTxPower = txPower;
#if defined( USE_BAND_915_HYBRID )
                        ChannelsMask[0] = channelsMask[0] & 0x00FF;
                        ChannelsMask[1] = channelsMask[1] & 0x0000;
                        ChannelsMask[2] = channelsMask[2] & 0x0000;
                        ChannelsMask[3] = channelsMask[3] & 0x0000;
                        ChannelsMask[4] = channelsMask[4] & 0x0001;
                        ChannelsMask[5] = channelsMask[5] & 0x0000;
#else
                        **ChannelsMask[0] = channelsMask[0];**
                        ChannelsMask[1] = channelsMask[1];
                        ChannelsMask[2] = channelsMask[2];
                        ChannelsMask[3] = channelsMask[3];
                        ChannelsMask[4] = channelsMask[4];
                        ChannelsMask[5] = channelsMask[5];
#endif

which overwrites the ChannelsMask[0] with whatever is in channelsMask[0] (in my case this is usually 15).

Should that #else not be #elif defined( USE_BAND_915 ).  I think this code might only be relevant to the US version.

mluis1, 2016-03-15:
The LINK_ADR_REQ MAC command is used to change the channel mask. The implementation is correct as is. 

Please note the upper case and lower case letters in the variables names.
**c**hannelsMask[0] = chMask;
**C**hannelsMask[0] = **c**hannelsMask[0];

The coding convention in use on this project are that global variables start by a upper case letter while local variables start by a lower case letter.

coljander, 2016-03-15:
Hi

Thanks.

I see my mistake.  I’ll check what is coming back from the gateway as at the end of that routine my ChannelMask[0] is only set to 15 so I’m only able to use 4 of the available channels.

Colin

https://github.com/Lora-net/LoRaMac-node/issues/55#issuecomment-196746841

closed 2016-03-21 by mluis1