Lora-net/LoRaMac-node #62: uplink delay after receiving a downlink message

pdemil, 2016-04-02:
Testing release 4.1.0: 
after receiving a downlink message (ADR ACK, ACK, device status request, ...) there seems to be a delay of 5 minutes before the next uplink is transmitted, when uplink messages are scheduled to be transmitted every 3 minutes.

Using the same application with release 4.0.0, we don't have this issue.

Someone else noticed the same?

mluis1, 2016-04-04:
Thanks for the report.
Could you please provide more information on the downlink message(s) that you send to the end-device?

If possible it would be nice to have the frame contents. Example:
Downlink frame:
60 F5 EC EE 00 20 00 00 28 14 E2 CC

pdemil, 2016-04-08:
Configuration: ABP, ADR on, APP_TX_DUTYCYCLE 180000000, no ACK, dutycycle on, 3 default channels.
The downlink messages are coming from the LoRaWAN server.

1-> 21:55:57.176 unconfirmedDataUpmessage, SF12, 
0<- 21:55:58.176 ConfirmedDataDown Mac (hex) : 035107000006
MAC.Command.LinkADRReq
MAC.Command.LinkADRReq.DataRate_TXPower : 0x51
MAC.Command.LinkADRReq.DataRate_TXPower.DataRate : 5
MAC.Command.LinkADRReq.DataRate_TXPower.TXPower : 1
MAC.Command.LinkADRReq.ChMask : 0x0007
MAC.Command.LinkADRReq.Redundancy : 0x00
MAC.Command.LinkADRReq.Redundancy.ChMaskCntl : 0
MAC.Command.LinkADRReq.Redundancy.NbRep : 0
MAC.Command.DevStatusReq
_8 minutes and 24 seconds later:_
2-> 22:04:22.621 UnconfirmedDataUp Data+mac, SF7, Mac (hex) : 030706fe24
MAC.Command.LinkADRAns
MAC.LinkADRAns.Status : 0x07
MAC.LinkADRAns.Status.ChannelMaskAck : 1
MAC.LinkADRAns.Status.DataRateAck : 1
MAC.LinkADRAns.Status.PowerAck : 1
MAC.Command.DevStatusAns
MAC.DevStatusAns.Battery : 254
MAC.DevStatusAns.Margin : 36

3-> 22:07:23.228 UnConfirmedDataUp Data, SF7
4-> 22:10:25.116 UnConfirmedDataUp Data, SF7
5-> 22:13:25.444 UnConfirmedDataUp Data, SF7

...continues, every 3 minutes...

19->22:55:25.595 UnConfirmedDataUp Data, SF7
20->22:58:25.160 UnConfirmedDataUp Data, SF7
1<-  22:58:25.160 ConfirmedDataDown Mac (hex) : 06
MAC.Command.DevStatusReq
_8 minutes and 25 seconds later_
21-> 23:06:50.811 UnconfirmedDataUp Data+Mac
MType : UnconfirmedDataUp
Flags : ADR : 1, ADRAckReq : 0, ACK : 1
Mac (hex) : 06fe1d
MAC.Command.DevStatusAns
MAC.DevStatusAns.Battery : 254
MAC.DevStatusAns.Margin : 29

djaeckle, 2016-04-13:
Hi pdemil,

I wasn't able to reproduce the issue yet.

1. Are you testing with the sample applications?
2. Does this issue appear every time or sometimes?

pdemil, 2016-05-12:
issue appears every time, but I will try again in v4.2. Closed.

closed 2016-05-12 by pdemil