Lora-net/LoRaMac-node #63: maximum payload messeges in EU ISM band

pBiczysko, 2016-04-11:
Hello, 
I am trying to send maximum payload messages (51 bytes on DR0 SF12BW125) from gateway to end-node and they are not received properly. 
My understanding is that if there are no fOpts to be transmitted the maximum payload for the message on DR0 is 51 bytes so adding the minimal FHDR, FPort, MHDR and MIC it equals to 64 bytes. When I try to send message with 51 bytes payload size on the packet forwarder log is exactly 64 bytes but the transmission does not even trigger the Rx interrupt on the end-node. To make it successful I have to reduce the payload to the size that makes the whole message size (packet forwarder log) 51 bytes. 

Another solution to this problem is to change the MaxPayloadOfDatarate value to 64 but I believe it is against LoRaSpec. 

Are my calculations wrong or is there some other variables that limit my payload size?

mluis1, 2016-04-11:
Thanks for the report. At first sight it looks like you are right. We will investigate the issue and try to provide a patch for the next release

pBiczysko, 2016-04-12:
One update, I run some tests and the issue is not only unique for the EU band I got the same results with US Band.

mluis1, 2016-04-12:
Yes, we are aware of it. Actually it applies to all supported bands. In the coming hours a patch should be available on the develop branch.

closed 2016-04-15 by mluis1