Lora-net/LoRaMac-node #65: Tx-Cw app example

dbanello, 2016-04-13:
I have two questions about Tx-Cw app example for Sx1272 at 868MHz (I've taken main.c example for SK-iM880A).
1) I've changed TX_OUTPUT_POWER  from 20 dBm to 14 dBm but after this modification SX1272 still transmits at 20dB
2) I've seen the following register settings:
    Radio.Write( 0x4B, 0x7B );
    Radio.Write( 0x3D, 0xAF );
    Radio.Write( 0x1e, 0x08 );
    Radio.Write( 0x4C, 0xC0 );
    Radio.Write( 0x4D, 0x03 );
    Radio.Write( 0x5A, 0x87 );
    Radio.Write( 0x63, 0x60 );
    Radio.Write( 0x01, 0x83 );

Is there some application note or documentation that describe these registers?

mluis1, 2016-04-13:
These registers description aren't disclosed by Semtech. Some of them are test registers which may damage the chip if their values are changed.

The current example sets the radio to always output 20 dBm regardless of the TX_OUTPUT_POWER value.

If you would like to test at another output power you may set the chip in FSK mode which is easier to setup in continuous wave mode and the results will be equivalent.

To do so you must set the radio in transmission mode using the FSK modem with a frequency deviation of 0 KHz.

Please find below an example: (I didn't test it but I think it should work)
```C
#include <string.h>
#include "board.h"
#include "radio.h"

#define RF_FREQUENCY                                868000000 // Hz
#define TX_OUTPUT_POWER                             14        // 14 dBm

#define FSK_FDEV                                    0         // Hz
#define FSK_DATARATE                                50e3      // bps
#define FSK_BANDWIDTH                               50e3      // Hz
#define FSK_AFC_BANDWIDTH                           83.333e3  // Hz
#define FSK_PREAMBLE_LENGTH                         5         // Same for Tx and Rx
#define FSK_FIX_LENGTH_PAYLOAD_ON                   true

int main( void )
{
    // Target board initialisation
    BoardInitMcu( );
    BoardInitPeriph( );

    // Radio initialization
    Radio.Init( NULL );

    Radio.SetChannel( RF_FREQUENCY );

    Radio.SetTxConfig( MODEM_FSK, TX_OUTPUT_POWER, FSK_FDEV, 0,
                                  FSK_DATARATE, 0,
                                  FSK_PREAMBLE_LENGTH, FSK_FIX_LENGTH_PAYLOAD_ON,
                                  true, 0, 0, 0, 3000000 );

    TimerInit( &Led4Timer, OnLed4TimerEvent ); 
    TimerSetValue( &Led4Timer, 90000 );

    TimerInit( &Led2Timer, OnLed2TimerEvent ); 
    TimerSetValue( &Led2Timer, 90000 );

    TimerInit( &Led3Timer, OnLed3TimerEvent ); 
    TimerSetValue( &Led3Timer, 90000 );
               
    // Switch LED 1 ON
    GpioWrite( &Led4, 0 );
    TimerStart( &Led4Timer );
    
    // Sets the radio in Tx mode
    Radio.Send( NULL, 0 );

    // Blink LEDs just to show some activity
    while( 1 )
    {
        if( Led4TimerEvent == true )
        {
            Led4TimerEvent = false;
            
            // Switch LED 4 OFF
            GpioWrite( &Led4, 4 );
            // Switch LED 2 ON
            GpioWrite( &Led2, 0 );
            TimerStart( &Led2Timer );
        }

        if( Led2TimerEvent == true )
        {
            Led2TimerEvent = false;
            
            // Switch LED 2 OFF
            GpioWrite( &Led2, 1 );
            // Switch LED 3 ON
            GpioWrite( &Led3, 0 );
            TimerStart( &Led3Timer );
        }
    
        if( Led3TimerEvent == true )
        {
            Led3TimerEvent = false;
            
            // Switch LED 3 OFF
            GpioWrite( &Led3, 1 );
            // Switch LED 1 ON
            GpioWrite( &Led4, 0 );
            TimerStart( &Led4Timer );
        }    
    }
}
```

dbanello, 2016-04-14:
Thank you very much. It works perfectly.
I don't know if here is the right place but I have another question:
Is a way to configure SX1272 in order to transmit a PN9 sequence?
Br

mluis1, 2016-04-14:
The best place to ask this kind of questions is to ask them on the [LoRa Community Semtech](https://semtech.force.com/lora) portal.
 
The SX1272 and SX1276 have no hardware PN9 sequence generator.
The drivers furnished in this repository don't enable the sending of PN9 sequence.

There would be a possibility which consists on using the Radio "Continuous Mode" chapter 4.2.13 of SX1272 datasheet and let a microcontroller generate the PN9 sequence and stream it to the DIO2/DATA and DIO1/DCLK radio pins.

dbanello, 2016-04-14:
Ok. Thank you very much for your answers.
Br

closed 2016-04-14 by dbanello