Lora-net/LoRaMac-node #66: SX127xSetOpMode improvement

JiapengLi, 2016-04-15:
Below is the implementation of SetOpMode, static variable opModePrev is used to keep previous mode, but this value is not always synced with the acctual mode, like situation user transmits one packet, when finished, SX127x will enter standby mode automatically, but opModePrev now is still in transmit mode, this will cause issue, if user doesn't call SetOpMode to set standby/sleep manually (actually not necessary for LoRa chip) before next transmitting, then the transmit will fail. 

Related patch is submitted but should have not solved the issue completely. @djaeckle
https://github.com/Lora-net/LoRaMac-node/commit/008f8dd414d779224c791bccf1ddbb8fc96963a7

Two possible solutions:
1. Abandon static local variable, use global value instead. Update the value when LoRa chip mode is 
changed internally automactically (TXDONE, RXDONE, and reset operation need sync the value to standby mode)
2. Abandon static local variable, read OpMode value from lora chip directly. This way costs some extra MCU resources but not much, should be more clean than previous one. 

``` C
void SX1276SetOpMode( uint8_t opMode )
{
    static uint8_t opModePrev = RF_OPMODE_STANDBY;

    if( opMode != opModePrev )
    {
        opModePrev = opMode;
        if( opMode == RF_OPMODE_SLEEP )
        {
            SX1276SetAntSwLowPower( true );
        }
        else
        {
            SX1276SetAntSwLowPower( false );
            if( opMode == RF_OPMODE_TRANSMITTER )
            {
                 SX1276SetAntSw( 1 );
            }
            else
            {
                 SX1276SetAntSw( 0 );
            }
        }
        SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode );
    }
}
```

mluis1, 2016-04-15:
Thanks @JiapengLi .

I'll add the below change to the next develop branch commit. (The same applies to SX1272)
```C
void SX1276SetOpMode( uint8_t opMode )
{
    if( opMode == RF_OPMODE_SLEEP )
    {
        SX1276SetAntSwLowPower( true );
    }
    else
    {
        SX1276SetAntSwLowPower( false );
        if( opMode == RF_OPMODE_TRANSMITTER )
        {
             SX1276SetAntSw( 1 );
        }
        else
        {
             SX1276SetAntSw( 0 );
        }
    }
    SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode );
}
```

JiapengLi, 2016-04-17:
Thank you, Miguel.

closed 2016-04-17 by JiapengLi