Lora-net/LoRaMac-node #67: CoIDE Toolchain

malroe, 2016-04-18:
Hi !
I am using the CoIDE toolchain in order to reprogram the iM880A node - ClassA project.

The toolchain is like:
CoIDE v. 1.7.8
GCC 4.9.3 - builds successfully
DFU File Manager v. 3.0.5 - for converting .hex to .dfu
DfuSe Demo v.3.0.5 - for uploading dfu to node, gives at Verify "Matching not good. First Difference at address 0x08000000: File byte is 0x00. Read Byte is 0xD0. !!!! 

BR
Michael Alrøe

mluis1, 2016-04-18:
For your information the CoIDE projects have been generated using version 2.0.1. I don't know if this has an influence.

Concerning the DFU file that you are trying to download, has it been generated with Target "LoRaMacClassA-868-Bootloader" selected on CoIDE?

The Target  "LoRaMacClassA-868-Bootloader" selects a different linker script and defines USE_BOOTLOADER. In order to move the start address from address 0x08000000 to address 0x08003000.

This means that the bootloader itself is placed at address 0x08000000 and its maximum size is 12 KB(0x3000 = 12288 decimal)

linker script file excerpt:
```C
/* Memory regions.*/ 
MEMORY
{
  FLASH (rx) : ORIGIN = 0x08003000, LENGTH = 128K - 0x3000
  RAM (rwx)  : ORIGIN = 0x20000000, LENGTH = 16K
}
```

board.c file excerpt:
```C
void BoardInitMcu( void )
{
    if( McuInitialized == false )
    {
#if defined( USE_BOOTLOADER )
        // Set the Vector Table base location at 0x3000
        SCB->VTOR = FLASH_BASE | 0x3000;
#endif
```

When using a bootloader the start address of your firmware should be 0x08003000 for the DFU update file. From your reporting it looks like that it starts at the beginning of the FLASH memory where the bootloader code itself should be located.

malroe, 2016-04-19:
Selecting target as "LoRaMacClassA-868-Bootloader" moved the firmware up to address 0x08003000 featuring succecfully DFU upload to mote.

Thank you :-)

closed 2016-04-19 by malroe

Smarco10, 2016-04-22:
hi all,

I reopen this topic because I have a linked issue that I do not still understand.
When I download the code from git (no modification) and I open the LoRaMote/Bootloader project, compilation goes wrong.

I work on Windows 10, with CoIDE v2.0.3 and an ARM toolchain: gcc-arm-none-eabi v4.9.3

Each time I compile the bootloader code it tells me that the "Bootloader.elf section '.text' will not fit in region 'FLASH'" "region 'FLASH' overflowed by 5212 bytes.
When I enter in src/boards/LoRaMote/cmsis/arm-gcc and look at the bootloader.ld its written that the length is 12k as explained 2-reply above.

Does anyone have the same issue? Do you  have any idea where does it come from?

I didn't give up on that and tried to change the size value (0x3000 to 0x4500) of the booloader (I know that I need to change all occurences to 0x3000 to the new size, that what I've done).
I succeed to compile it and flash it on the device with JTAG connector.
But when I press the SAR sensor and plug it on USB, Windows says that it does not recognize it  (despite the fact that I install the "stsw-link009 Driver". I have also test to connect it on another computer but I face the same problem.

Thank you for any help you may provide me.
Sincerly
Marc-Antoine MARTIN.

mluis1, 2016-04-22:
I have just tried to replicate what you are trying to do and everything went OK.

My setup is the following:
Windows 7, CoIDE v2.0.1 and gcc 4.8 2014q2
The DFU drivers that I use are DfuSe_Demo_V3.0.5_Setup.exe that you can find on ST web site.

Bellow you will find the tool compilation log output.

I have attached the compiler produced .hex file. Once downloaded you need to rename it from .txt to .hex
[BootLoader.txt](https://github.com/Lora-net/LoRaMac-node/files/231516/BootLoader.txt)

```
GCC HOME: C:\Program Files (x86)\GNU Tools ARM Embedded\4.8 2014q2\bin
compile:
    [mkdir] Created dir: C:\Temp\LoRaMac-node\coIDE\LoRaMote\Bootloader\bootloader\Debug\bin
    [mkdir] Created dir: C:\Temp\LoRaMac-node\coIDE\LoRaMote\Bootloader\bootloader\Debug\obj
       [cc] 41 total files to be compiled.
       [cc] arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -Wall -ffunction-sections -g -Os -std=c99 -c -DUSE_NO_TIMER -IC:\Temp\LoRaMac-node\src\boards\mcu\stm32\stm32_usb-fs-device_driver\inc -IC:\Temp\LoRaMac-node\src\system -IC:\Temp\LoRaMac-node\src\boards\mcu\stm32\cmsis -IC:\Temp\LoRaMac-node\src\boards\loramote -IC:\Temp\LoRaMac-node\src -IC:\Temp\LoRaMac-node\src\mac -IC:\Temp\LoRaMac-node\src\system\crypto -IC:\Temp\LoRaMac-node\src\boards\mcu\stm32\stm32l1xx_stdperiph_driver\inc -IC:\Temp\LoRaMac-node\src\radio -IC:\Temp\LoRaMac-node\src\peripherals -IC:\Temp\LoRaMac-node\src\boards\mcu\stm32 -IC:\Temp\LoRaMac-node\src\boards\loramote\cmsis -IC:\Temp\LoRaMac-node\src\boards\loramote\usb\dfu\inc C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_gpio.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32_USB-FS-Device_Driver\src\usb_core.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\sysIrqHandlers.c C:\Temp\LoRaMac-node\src\system\i2c.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_pwr.c C:\Temp\LoRaMac-node\src\peripherals\sx1509.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32_USB-FS-Device_Driver\src\usb_int.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_flash.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\cmsis\arm-gcc\startup_stm32l1xx_md.s C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb\dfu\src\usb_desc.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32_USB-FS-Device_Driver\src\usb_regs.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_exti.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\utilities.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_rcc.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb\dfu\src\usb_prop.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb\dfu\src\usb_pwr.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\cmsis\system_stm32l1xx.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_flash_ramfunc.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32_USB-FS-Device_Driver\src\usb_init.c C:\Temp\LoRaMac-node\src\apps\BootLoader\LoRaMote\main.c C:\Temp\LoRaMac-node\src\peripherals\gpio-ioe.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\i2c-board.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_adc.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb\dfu\src\dfu_mal.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb\dfu\src\flash_if.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\gpio-board.c C:\Temp\LoRaMac-node\src\system\gpio.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb-dfu-board.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_dbgmcu.c C:\Temp\LoRaMac-node\src\peripherals\sx9500.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_syscfg.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32_USB-FS-Device_Driver\src\usb_sil.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_dma.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_usart.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_spi.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_i2c.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\misc.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_rtc.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32L1xx_StdPeriph_Driver\src\stm32l1xx_tim.c C:\Temp\LoRaMac-node\src\boards\mcu\stm32\STM32_USB-FS-Device_Driver\src\usb_mem.c C:\Temp\LoRaMac-node\src\boards\LoRaMote\usb\dfu\src\usb_istr.c
       [cc] C:\Users\mluis\AppData\Local\Temp\cc9zNyti.s: Assembler messages:
       [cc] C:\Users\mluis\AppData\Local\Temp\cc9zNyti.s:18: Warning: ignoring changed section attributes for .data
       [cc] Starting link
       [cc] arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -g -nostartfiles -Wl,-Map=BootLoader.map -Os -Wl,--gc-sections --specs=nano.specs -LC:/Temp/LoRaMac-node/src/boards/loramote/cmsis/arm-gcc -Wl,-TC:/Temp/LoRaMac-node/src/boards/loramote/cmsis/arm-gcc/stm32l1xx_md_flash_bootloader.ld -g -o BootLoader.elf ..\obj\stm32l1xx_gpio.o ..\obj\usb_core.o ..\obj\sysIrqHandlers.o ..\obj\i2c.o ..\obj\stm32l1xx_pwr.o ..\obj\sx1509.o ..\obj\usb_int.o ..\obj\stm32l1xx_flash.o ..\obj\startup_stm32l1xx_md.o ..\obj\usb_desc.o ..\obj\usb_regs.o ..\obj\stm32l1xx_exti.o ..\obj\utilities.o ..\obj\stm32l1xx_rcc.o ..\obj\usb_prop.o ..\obj\usb_pwr.o ..\obj\system_stm32l1xx.o ..\obj\stm32l1xx_flash_ramfunc.o ..\obj\usb_init.o ..\obj\main.o ..\obj\gpio-ioe.o ..\o [... cut ...] obj\gpio-board.o ..\obj\gpio.o ..\obj\usb-dfu-board.o ..\obj\stm32l1xx_dbgmcu.o ..\obj\sx9500.o ..\obj\stm32l1xx_syscfg.o ..\obj\usb_sil.o ..\obj\stm32l1xx_dma.o ..\obj\stm32l1xx_usart.o ..\obj\stm32l1xx_spi.o ..\obj\stm32l1xx_i2c.o ..\obj\misc.o ..\obj\stm32l1xx_rtc.o ..\obj\stm32l1xx_tim.o ..\obj\usb_mem.o ..\obj\usb_istr.o -lm
Program Size:
      text	   data	    bss	    dec	    hex	filename
     10764	   1136	   1340	  13240	   33b8	BootLoader.elf

BUILD SUCCESSFUL
Total time: 16 seconds
```

Smarco10, 2016-04-22:
Thank you for your quick reply, I'll try to reproduce your result (by changing toolchain and see if anything change).
I uploaded you hex file and the behavior changed, now it seems working well. 3 LED blink and the driver seems recognize the LoRaMote as well as the DFuSeDemo see the device.
I may use DFuse to upgrade the firmware, but weirdly, it doesn't seem to be started by the bootloader (I didn't change anything in the ".ld" files) neither I change the memory offset. Firmware work very well when I flash it through JTAG connector at 0x80000000.
I don't know if the toolchain is the guilty thing but I'll try with your version of ARM crosscompile toolchain (v4.8 2014q2).

Thank you for your help, I'll keep you up with lastest news.
Sincerely,
Marc-Antoine MARTIN

Smarco10, 2016-04-28:
Hi Luis,

I'm back on this topic, just to let you know that I succeed to compile and use the bootloader.
The problem came from GCC params passed at compilation time.
I don't know why, my project was configured on debug information level 2 (-g2) which keep the code lines into the output program in order to run a debugger. Yours is wonfigured with no debug (-g or -g0).

The second parameter which wasn't set on my project was "-ffunction-sections" which, as well I understood what it does, shrink the output binary during linking edition with removing unused function. 

It results in a smaller binary which with these both options is small enough. 

Hiopping it will help others who has same problem.
Sincerely