Lora-net/LoRaMac-node #68: About low level RF switch control apis

JiapengLi, 2016-04-19:
SX1276.RxTx variable is not synced when the device is set to low power mode.
This may cause RxWindow2 receive sesitivity lost issue of LoRaWAN stack.

    // Reproduce the issue
    SX1276SetAntSwLowPower(false);
    SX1276SetAntSw(0);      // SX1276.RxTx is 0 now
    SX1276SetAntSwLowPower(false);
    SX1276SetAntSw(0);      // SX1276.RxTx is still 0, this function is not worked as expected

The related functions locate at sx127x-board.c files.

``` c
void SX1276SetAntSwLowPower( bool status )
{
    if( RadioIsActive != status )
    {
        RadioIsActive = status;
    
        if( status == false )
        {
            SX1276AntSwInit( );
        }
        else
        {
            SX1276AntSwDeInit( );
        }
    }
}

void SX1276AntSwInit( void )
{
    GpioInit( &AntSwitchLf, RADIO_ANT_SWITCH_LF, PIN_OUTPUT, PIN_PUSH_PULL, PIN_PULL_UP, 1 );
    GpioInit( &AntSwitchHf, RADIO_ANT_SWITCH_HF, PIN_OUTPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
}

void SX1276AntSwDeInit( void )
{
    GpioInit( &AntSwitchLf, RADIO_ANT_SWITCH_LF, PIN_OUTPUT, PIN_OPEN_DRAIN, PIN_NO_PULL, 0 );
    GpioInit( &AntSwitchHf, RADIO_ANT_SWITCH_HF, PIN_OUTPUT, PIN_OPEN_DRAIN, PIN_NO_PULL, 0 );
}

void SX1276SetAntSw( uint8_t rxTx )
{
    if( SX1276.RxTx == rxTx )
    {
        return;
    }

    SX1276.RxTx = rxTx;

    if( rxTx != 0 ) // 1: TX, 0: RX
    {
        GpioWrite( &AntSwitchLf, 0 );
        GpioWrite( &AntSwitchHf, 1 );
    }
    else
    {
        GpioWrite( &AntSwitchLf, 1 );
        GpioWrite( &AntSwitchHf, 0 );
    }
}
```

nestorayuso, 2016-04-19:
why to manage the RF Switch control pin with a GPIO pin and not with the SX127x RXTX dedicated pin?

jmalangoni, 2016-04-19:
>>SX1276SetAntSw(0); // SX1276.RxTx is still 0, this function is not worked as expected

to set to tx, you must pass non-zero as argument. you are setting to rx from rx i think.

mluis1, 2016-04-19:
@nestorayuso It is is for flexibility reasons. Depending on the antenna switch used the polarity may be inverted (requires additional external inverter), some may need to have 2 control pins. Globally it is PCB design dependent.

If your board design directly connects the Radio RxTx pin to the antenna switch then you could remove the contents of these functions.

The board files in this project are meant to be adapted to the specific board design.

closed 2016-05-12 by mluis1

SaiPrakash14, 2017-10-25:
I am having the SX1272 chip which have TX_switch and Rx_switch (https://www.rfsolutions.co.uk/radio-modules-c10/frequency-c57/fm-lora-transceiver-module-pre-set-to-868mhz-p4682).
but in code implementation they used single RXTX pin to control antenna operation(transmission or reception). based upon RXTX pin they control antenna operation
How can i modified this related to my situation (i am having the separate pins RX_Switch & TX_Switch).
How can I connect those pins to GPIO of micro controller(MCU)
Could you please help me to solve this problem

Thanks in Advance.

Regards
Ashok

nestorayuso, 2017-10-25:
please don't post the same questions in several places.

first you must create a folder inside boards with the name of your board.
the LoRaMote board is quite similar to your board,use it as a template.
copy LoRaMote files to your new folder.
looking at here: https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/LoRaMote/sx1272-board.c#L195 you can see it already uses two gpio pins to manage external RF switch, I think you need to modify nothing here.

Modify board.h https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/LoRaMote/board.h#L103 with your particular gpio pin asignations.