Lora-net/LoRaMac-node #70: Missing Sequence number when using ADR

shankuuuu, 2016-04-21:
I am using v4.1 stack for my Lora modem implementation.
I have enabled ADR and is using confirmed msgs.
I am using a kerlink gateway and Loriot server.

This is what I find when sending a lot of data to the server.

[image]

As you can see, when the data rate improves from SF12 to SF11 or SF11 to SF10, I am missing a sequence number at the server side received data.

Is this a node implementation issue or is it something to do with gateway/server?

Can anyone help me to find out why this is happening ?

rayaan92, 2016-04-25:
I have been experiencing the exact same problem, where I am missing sequence numbers

petrussdutoit, 2016-04-26:
Hi,

The sequence number is not actually missing. What is happening is the node
is responding with an empty frame with a LinkADRAns MAC command in the
FOpts field  in response to the LinkADRReq MAC command that was sent in the
previous downlink frame. Because the response frame has a 0 length data
payload and is sent on port 0 the server is not showing it as data.

I have seen this behavior with both the Semtech GitHub stack as well as the
Microchip RN2483 module.

Regards
Petrus Du Toit

closed 2016-05-12 by mluis1