Lora-net/LoRaMac-node #71: Notification If the ACK is not received on ConfirmedFrame

FoujanetG, 2016-04-22:
Hi,

If I send a confirmedFrame, the LoraWan stack notifies the user side with McpsConfirm and McpsIndication functions.

But if the acknowledge is not received, the user side is not notifies. I would like to know if the acknowloedge is received or not.

the OnAckTimeoutTimerEvent is called after each confirmedframe without ACK returned.

Do you have an solution ?

djaeckle, 2016-04-25:
Hi FoujanetG,

thanks for the report. In that case, the stack notifies the upper layer with the `McpsConfirm` callback. The parameter list reports:

`Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT`
`AckReceived = false`

However, please notice, that the stack provides the notification not until all frames has been sent, according to the number of trials in `McpsReqConfirmed_t`. In the sample application, this parameter is set to 8.

daniwebCH, 2016-04-25:
Hi Daniel,

I also notice that I do not get the described callback...
what in case you do not wish retransmissions and you set NbTrials to 1 ?

Daniele

djaeckle, 2016-04-25:
Hi Daniele,

I couldn't reproduce the issue yet. Could you provide some more details?

daniwebCH, 2016-04-25:
I do not know which details you are more interested in, I give you some 
inputs.

I use the base of LoRaMote->LoRaMac->ClassA version 4.1.
I send a frame ADR enabled, DR_0, 20 bytes payload to port 11, with 
confirmation request and NbTrials set to 1 ( I expect that are no 
repetitions therefore dutyCycle should not play a Role).
I see that OnRadioTxDone function is called.
After 20 seconds I expect that I should receive the ACK and my upper 
Layer declares this as not success.

I use Actility as network provider (in case I could also try it with the 
demo server of Actility or with the Semtech RaspberryPi box got at bootcamp)

djaeckle, 2016-04-25:
Thanks for the feedback. Does the stack not provide the callback `McpsConfirm`? Or is the problem that the node does not receive an ACK?

daniwebCH, 2016-04-25:
The problem is that the callback McpsConfirm is not called in case of 
the node did not receive the ACK (I probably confused you by specifying 
the network provider that should not have an impact).

FoujanetG, 2016-04-25:
Hi,

Thanks a lot djaeckle.
The McpsConfirm callback is called after a SendConfirm ( after all trials or when the ACk is received)
If the Ack is received, 
```
Status = LORAMAC_EVENT_INFO_STATUS_OK
AckReceived = true
```

and if the ACK is not received, 
```
Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT
AckReceived = false
```

djaeckle, 2016-04-29:
Hi daniwebCH,

I'm still not able to reproduce the issue. Could you provide more details on the implementation of your application?

closed 2016-06-03 by mluis1

JulienArmenti, 2016-11-08:
Hi,

I have a similar problem:

I do an OTA activation, it's work correctly and I can send data periodicaly.

But, In this case:
 I do the OTA activation, I send data (one or more packet(s)) and I stop the send to transmit data in LoRa (without using the LoRaWAN stack). After this, I return in LoRaWAN and I send packet periodicaly and I have always this status:

Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT
AckReceived = false

I have tried with confirmed and unconfirmed data, the problem is the same. 
If I restart the OTA activation when I return in LoRaWAN confiuration, I have no problem.
Sometimes I can see a receive packet in the gateway side but I never had a confirmation in the mote side

I use the master branch v1.01 with a semtech gateaway IOTkit
If you have an idea about this problem...

Thanks
J.A

djaeckle, 2016-11-11:
Hi JulienArmenti,

it seems like this issue is not related to this thread directly. Could you please open up a new issue? Could you provide some more details? Do you reconfigure the radio at some place?

singh667, 2020-07-30:
hi sir my mcpconfirmation callback is okk but when i check the status  then status is comming like that i debug all my possible way but i did not able to find where is the problem why rx2 timerout  
Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT
AckReceived = false

i am not understanding one thing when i send the confirmed up message and formate of the confirmed message u
[image]
[image]
[image]

this file complete data what i send to the server and what downlink data i get 
please consider where is the problem

singh667, 2020-07-30:
i think some problem to server  downlink data i received there some parameter value is the mismatch