Lora-net/LoRaMac-node #75: SNR value for MAC_DEV_STATUS_ANS

Mobiak, 2016-04-26:
In file sx1272.c when device receive a message, it use snr value only to calculate the RSSI. The value transmitted by callback to LoRaMac is the raw value directly read from the SX1272 register.
But in the stack LoRaWan, there is no transformation of this value, it is stored and given back to the network layer as is.

Issue #29 is telling that it should transfer the raw data value, but the LoRaWAN Specification 1R0  and the LoRaWAN1.0.1final_876_3 are both saying:

_"The margin (Margin) is the demodulation signal-to-noise ratio in dB rounded to the nearest integer value for the last successfully received DevStatusReq command.  It is a signed integer of 6 bits with a minimum value of -32 and a maximum value of 31."_

So, should the LoRaWan layer implement the transformation to get the refined value on 6 bit in order to enforce LoRaWan specification ?

sx1272.c - function SX1272OnDio0Irq( ) : reading the value and transformed it for rssi calculation
```

        SX1272.Settings.LoRaPacketHandler.SnrValue = SX1272Read( REG_LR_PKTSNRVALUE );
        if( SX1272.Settings.LoRaPacketHandler.SnrValue & 0x80 ) // The SNR sign bit is 1
        {
            // Invert and divide by 4
            snr = ( ( ~SX1272.Settings.LoRaPacketHandler.SnrValue + 1 ) & 0xFF ) >> 2;
            snr = -snr;
        }
        else
        {
            // Divide by 4
            snr = ( SX1272.Settings.LoRaPacketHandler.SnrValue & 0xFF ) >> 2;
        }

        int16_t rssi = SX1272Read( REG_LR_PKTRSSIVALUE );
        if( snr < 0 )
        {
            SX1272.Settings.LoRaPacketHandler.RssiValue = RSSI_OFFSET + rssi + ( rssi >> 4 ) +
                                                          snr;
        }
        else
        {    
            SX1272.Settings.LoRaPacketHandler.RssiValue = RSSI_OFFSET + rssi + ( rssi >> 4 );
        }

        SX1272.Settings.LoRaPacketHandler.Size = SX1272Read( REG_LR_RXNBBYTES );
        SX1272ReadFifo( RxBuffer, SX1272.Settings.LoRaPacketHandler.Size );
    
        if( SX1272.Settings.LoRa.RxContinuous == false )
        {
            SX1272.Settings.State = RF_IDLE;
        }
        TimerStop( &RxTimeoutTimer );

        if( ( RadioEvents != NULL ) && ( RadioEvents->RxDone != NULL ) )
        {
            RadioEvents->RxDone( RxBuffer, SX1272.Settings.LoRaPacketHandler.Size, SX1272.Settings.LoRaPacketHandler.RssiValue, SX1272.Settings.LoRaPacketHandler.SnrValue );
        }

```

LoRaMac.c - function OnRadioRxDone( ) : prototype

`        static void OnRadioRxDone( uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr )`

LoRaMac.c - function OnRadioRxDone( ) : store the value

`        LoRaMacEventInfo.RxSnr = snr;`

LoRaMac.c - function LoRaMacProcessMacCommands( ) : use the previously stored value
```

        case SRV_MAC_DEV_STATUS_REQ:
            {
                uint8_t batteryLevel = BAT_LEVEL_NO_MEASURE;
                if( ( LoRaMacCallbacks != NULL ) && ( LoRaMacCallbacks->GetBatteryLevel != NULL ) )
                {
                    batteryLevel = LoRaMacCallbacks->GetBatteryLevel( );
                }
                AddMacCommand( MOTE_MAC_DEV_STATUS_ANS, batteryLevel, LoRaMacEventInfo.RxSnr );
            }
            break;
```

mluis1, 2016-04-26:
We think that the version 4.1 correctly implements the SNR value reporting. 

The MAC command reports the radio register raw value as specified in the LoRaWAN specification. 

For your information the radio register value ranges from -32 to 31. To get the real value you must dived it by 4.

```C
static void ProcessMacCommands( uint8_t *payload, uint8_t macIndex, uint8_t commandsSize, uint8_t snr )
{
    while( macIndex < commandsSize )
    {
        // Decode Frame MAC commands
        switch( payload[macIndex++] )
        {
...
            case SRV_MAC_DEV_STATUS_REQ:
                {
                    uint8_t batteryLevel = BAT_LEVEL_NO_MEASURE;
                    if( ( LoRaMacCallbacks != NULL ) && ( LoRaMacCallbacks->GetBatteryLevel != NULL ) )
                    {
                        batteryLevel = LoRaMacCallbacks->GetBatteryLevel( );
                    }
                    AddMacCommand( MOTE_MAC_DEV_STATUS_ANS, batteryLevel, snr );
                    break;
                }
...
}

...

static void OnRadioRxDone( uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr )
{
...
    McpsConfirm.AckReceived = false;
    McpsIndication.Rssi = rssi;
    McpsIndication.Snr = snr;
    McpsIndication.RxSlot = RxSlot;
    McpsIndication.Port = 0;
    McpsIndication.Multicast = 0;
    McpsIndication.FramePending = 0;
    McpsIndication.Buffer = NULL;
    McpsIndication.BufferSize = 0;
    McpsIndication.RxData = false;
    McpsIndication.AckReceived = false;
    McpsIndication.DownLinkCounter = 0;
    McpsIndication.McpsIndication = MCPS_UNCONFIRMED;
...
                    if( fCtrl.Bits.FOptsLen > 0 )
                    {
                        // Decode Options field MAC commands
                        ProcessMacCommands( payload, 8, appPayloadStartIndex, snr );
                    }
                    if( ( ( size - 4 ) - appPayloadStartIndex ) > 0 )
                    {
                        port = payload[appPayloadStartIndex++];
                        frameLen = ( size - 4 ) - appPayloadStartIndex;

                        McpsIndication.Port = port;

                        if( port == 0 )
                        {
                            LoRaMacPayloadDecrypt( payload + appPayloadStartIndex,
                                                   frameLen,
                                                   nwkSKey,
                                                   address,
                                                   DOWN_LINK,
                                                   downLinkCounter,
                                                   LoRaMacRxPayload );

                            // Decode frame payload MAC commands
                            ProcessMacCommands( LoRaMacRxPayload, 0, frameLen, snr );
                        }
                        else
                        {
...
}
```
Which version are you based on?
Are you using the V3.0 wrapper?

From the code examples you provide it looks like that you are using the V3.0 wrapper. The LoRaMacEventInfo.RxSnr variable doesn't exist anymore on the LoRaMac.c file.

Please also note that as stated in the [LoRaMAC node Wiki](https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Wiki) page the support for the wrapper 3.0 will be removed starting at version 5.0.0

Could you please close Issue #29 again. For clarity reasons one shouldn't create duplicated issues.

Mobiak, 2016-04-27:
Still the specification ask for a value signed on six bit, and the two others are reserved for future evolution (RFU). Currently the stack transfer the raw value signed on 8 bit, and so it overide the RFU space.

I agreed that to convert the raw value into the refined one is just need a division, however, it should be implemented in the stack and not on the LoRa server nor the application server.

I'm using version 4.0.0 and 4.1.0 on two different project, they will be upgraded with v5.0.0 as soon as it get available =)

Dully noted for the end of the wrapper v3 support.

mluis1, 2016-05-12:
Please find below the answer I got from the LoRa-Alliance technical committee.

"The LoRaWAN specification describes the format of the fields exchanged in the radio frames over the air.
It does not describe the API between the end-device’s MAC layer and the device’s Application layer.
If the MAC layer designer exposes an SNR value on 8 bits to the application, it is a perfectly valid choices, as long as it explains the format of the data presented.  He is free to perform any scaling/computation he wants on the the MAC field before presenting them to the application if he thinks that makes more sense."

Mobiak, 2016-05-12:
Thanks for the answer.

closed 2016-05-12 by Mobiak