Lora-net/LoRaMac-node #76: Why I cann't receive any packet from RF?

wuylz, 2016-04-29:
Hi,
I use iC880A as a concentrator,  iM880A as nodes.
the concentrator is running "basic_packet_forward" program, and the node is running the "LoRaMac-node" code.
I have defined OVER_THE_AIR_ACTIVATION 1   to using OTAA,  but when I start the node, the concentrator cann't receive any packet, why?
the rasberry Pi printf message is : 
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
#### END #####

Thanks

rossihwang, 2016-05-05:
you should make sure they are in same frequency plan, and no other environmental interference

mluis1, 2016-05-12:
Can we close this issue?

closed 2016-05-31 by mluis1