Lora-net/LoRaMac-node #77: read value

Bruce9096, 2016-05-03:
Hi there

I use the 880a as the node, 880c as the concentrator, now trying to read the value of the GPIO pin, for example, GpioRead(PA_14), and Keil warning that" incompatible integer to pointer conversion passing 'int' to parameter of type of 'Gpio_t'

Could you please tell me how to figure this out? do I need to define the pin as Gpio_t format? could you please give the detail to make the read successfully,

thank you so much.

best wishes

rayaan92, 2016-05-03:
so if you have a look in the board.h file all the pins are defined there, and then initialized in the board.c file, please see the below snippet:

board.h:
#define TEST     PA_14

extern Gpio_t test;

board.c:
 Gpio_t test;

then initialize the pin in BoardInitPeriph:
GpioInit ( &test, TEST, PIN_INPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0);

then you can use
uint8_t read=0; 
read = GpioRead(&test);

Bruce9096, 2016-05-03:
Hi 

Thank you so much for your kindness help，thank you so much^_^

best wishes

Linyu

mluis1, 2016-05-12:
I close the issue as it looks like it is solved.

closed 2016-05-12 by mluis1

rayaan92, 2016-05-24:
> Hi Rayaan
>
> I am now trying to connect a I2C interface temperature sensor to Im880A, I
> have checked the src lib has I2C.c and .h file, could you please explain
> that how to use these functions?
>
> could you please give me a simple example, thank you very much.
>
> Best wishes
>
> Linyu cai

Hi Linyu cai,

The LoraMote and the SK-im880a are both based on the Lora module from IMST,
so you can check in the loramote project, there are a few sensors that is
interfaced to the module using the i2c serila interface, you can use that
as a base guideline, and if you require any further assistance oplease feel
free to drop me a mail any time and I will gladly assist. I am currently
working on setting up the SPi2 bus, ran into some issues but once i get it
working I will gladly share it with you.

Kind Regards,
And everything of the best
Rayaan Latief

Bruce9096, 2016-05-25:
Hi Rayaan Latief

Thank you for your kindness reply, I will check the Loramote project.

Again, thank you so much, (as I am a fresh man to learn this part, maybe
some of the questions are stupid, so still thank you for your patience.)

best wishes

Linyu cai

rayaan92, 2016-05-25:
It's not a problem man, I am here to help you with anything not matter how
stupid you think it may sound. everything of the best with your project I
am also going to use the i2c interface  to test some sensors so if you get
stuck don't hesitate to ask.

everything of the best again.
Kind regards,
Rayaan

Bruce9096, 2016-05-31:
Hi Rayaan
Good morning.
May I ask you whether you do any coding on im871 M-Bus？

regards

linyu cai