Lora-net/LoRaMac-node #79: 600 Baud rate

rayaan92, 2016-05-10:
Good I urgently need assistance with configuring the uart baud rate to 600 on the SK-IM880a,
any assistance is highly appreciated Thank you in advance.

mluis1, 2016-05-12:
I'll assume you want to set the baudrate at 600 bit/s.

```C
UartInit( &Uart1, UART_1, UART_TX, UART_RX );
UartConfig( &Uart1, RX_TX, 600, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );
```

rayaan92, 2016-05-12:
It's okay I got it working, in order to get a baud rate of 600 I had to
change the APB2 clock divided to 2. Because on the default clock settings
the lowest baud rate is 1200 according to the stm32l151cbu6 datasheet.

mluis1, 2016-05-12:
Nice, I'll then close the issue.

closed 2016-05-12 by mluis1

rayaan92, 2016-05-12:
I can attach snippets of the code for you.

mluis1, 2016-05-12:
Yes, it maybe useful for others.

rayaan92, 2016-05-12:
to get a uart baud rate of 600, I did the following:

in the system_stm32l1xx.c file:

line 348,I change the APB2 clock divder value to 2
 RCC->CFGR |= (uint32_t)RCC_CFGR_PPRE2_DIV2;

and when I wrote to the uart is did the following:

__disable_irq( );
while( check == 0)
{
cs5480_register_initialization( vgain, igain);
DelayMs(500);
check = 1;
}
__enable_irq( );