Lora-net/LoRaMac-node #82: Communication problem SensorNode with SIM39EA since HAL update

MichaelDevid, 2016-05-18:
Hi,

Thanks for the great HAL update. Only I now have a problem at line 307 in gps.c 

`uint8_t GpsParseGpsData( int8_t *rxBuffer, int32_t rxBufferSize )
{
    uint8_t i = 1;
    uint8_t j = 0;
    uint8_t fieldSize = 0;
    
    if( rxBuffer[0] != '$' )
    {
        GpsMcuInvertPpsTrigger( );
        return FAIL;
    }
`

Since I updated It fails at this check when communicating with the SIM39EA on the SensorNode board. Do you know a possible solution?

mluis1, 2016-05-18:
Thanks for the report. We will investigate this as soon as possible and provide a patch in the develop branch.

MichaelDevid, 2016-05-24:
I was looking at the problem again and it seems that now the program fails at line 281 in gps.c most of the time. It seems as if the stack is losses some bytes.

MichaelDevid, 2016-06-03:
Any news on this issue?

mluis1, 2016-06-07:
I think that I have found the issue.

Sometime ago we had to update the GPS module on the LoRaMote platform. The only difference between the old and new module is that the PPS signal is inverted. That's why we had to add the ``` GpsMcuInvertPpsTrigger( ) ``` function.

While updating to the new HAL we have applied by mistake the same fix to the SensorNode platform.

Right now I don't have access to a SensorNode platform. Could you please try the following change to ```src\boards\SensorNode\gps-board.c``` file.
Original:
```C
void GpsMcuInvertPpsTrigger( void )
{
    if( PpsTigger == PpsTiggerIsRising )
    {
        PpsTigger = PpsTiggerIsFalling;
        GpioSetInterrupt( &GpsPps, IRQ_FALLING_EDGE, IRQ_VERY_LOW_PRIORITY, &GpsMcuOnPpsSignal );
    }
    else
    {
        PpsTigger = PpsTiggerIsRising;
        GpioSetInterrupt( &GpsPps, IRQ_RISING_EDGE, IRQ_VERY_LOW_PRIORITY, &GpsMcuOnPpsSignal );
    }
}
```
Changed to:
```C
void GpsMcuInvertPpsTrigger( void )
{

}
```

MichaelDevid, 2016-06-07:
Problem solved! Thank you very much!

MichaelDevid, 2016-06-08:
It seems to have solved the issue but my current hardware isn't optimal so I will check with new hardware next week.

closed 2016-07-29 by mluis1