Lora-net/LoRaMac-node #85: Two or more timers eventually stop

ChrisEAlfred, 2016-05-25:
I wrote a simple test application that starts 2 timers. One timer is 500ms the second is 1000ms. They are started, and when elapsed, they are stopped and restarted.

After a random amount of time, but usually within 10s, the timers both stop - it is as though the TimeStart routine does not start timers any more due to a race condition.

If only one timer is running, all works correctly.

I have attached the code. The BoardDebug() is a custom routine prints debug via serial port.

This code is running on an STM32F411 Nucleo board. Admittedly, the HAL low power is different on the F411, so low power is disabled for now.

#include <string.h>
#include <math.h>
#include "board.h"

#define USE_BOTH_TIMERS     1

/*!
 * \brief Function executed on Test 1 Timeout event
 */
static TimerEvent_t Test1Timer;
static bool test1TimerEvent = false;
static void OnTest1TimerEvent( void )
{
    // NOTE: executed in interrupt space
    TimerStop( &Test1Timer );
    test1TimerEvent = true;
}

#if USE_BOTH_TIMERS
/*!
 * \brief Function executed on Test 2 Timeout event
 */
static TimerEvent_t Test2Timer;
static bool test2TimerEvent = false;
static void OnTest2TimerEvent( void )
{
    // NOTE: executed in interrupt space
    TimerStop( &Test2Timer );
    test2TimerEvent = true;
}
#endif

/**
 * Main application entry point.
 */
int main( void )
{
    int result;
    uint32_t lastHeartbeatTick;

    BoardInitMcu( );
    BoardInitPeriph( );

    BoardDebug("Timers\r\n------\r\n");

    // Setup timers
    TimerInit( &Test1Timer, OnTest1TimerEvent );
    TimerSetValue( &Test1Timer, 500 );
    TimerStart( &Test1Timer );

#if USE_BOTH_TIMERS
    TimerInit( &Test2Timer, OnTest2TimerEvent );
    TimerSetValue( &Test2Timer, 1000 );
    TimerStart( &Test2Timer );
#endif

    while( 1 )
    {
        // Heartbeat
        if (HAL_GetTick() - lastHeartbeatTick >= 1000)
        {
            lastHeartbeatTick = HAL_GetTick();
            GpioToggle(&Led4);
            BoardDebug("[Beat]\r\n");
        }

        if (test1TimerEvent)
        {
            test1TimerEvent = false;
            result = TimerStart( &Test1Timer );
            BoardDebug("Test1Timer %04X\r\n", result);
        }
#if USE_BOTH_TIMERS
        if (test2TimerEvent)
        {
            test2TimerEvent = false;
            result = TimerStart( &Test2Timer );
            BoardDebug("Test2Timer %04X\r\n", result);
        }
#endif
    }
}

ChrisEAlfred, 2016-05-25:
The last successful TimerStart call always executes timer.c :: TimerStart clause:

        if( obj->Timestamp < remainingTime )
        {
            TimerInsertNewHeadTimer( obj, remainingTime );
            result |= 0x20; // debug that we got here
        }

But it is not the first time that it ran this part of the code.

shankuuuu, 2016-05-31:
I am using STM32L073 uC. I also have disabled the low power mode.
I can see that after running well for sometime, the timer stops.
The time when timer stops is random. Sometimes it can run few hours, sometimes few mins.

ChrisEAlfred, 2016-06-01:
I now have an STM32L152RET6 Nucleo board (NUCLEO-L152RE) which is the closest I could get to the SensorNode which uses an STM32L151.

This means that I can use the exact same code and use ultra-low power mode.

The NUCLEO-L152RE also exhibits the timer stopping. I think this is a real bug in LoRa-net timer handling.

shakuuuu, were you running my example code? If you were running the normal LoRa examples, the probability is much lower which could explain why it took longer for it to happen on your board.

ChrisEAlfred, 2016-06-02:
Still testing, but it looks like if you schedule another timer event to occur at the same millisecond as the timer event on the head, then the timers stop.

It has nothing to do with low power.

shankuuuu, 2016-06-03:
@ChrisEAlfred  yes I have been using the Sensor Node code. In this application case,most of the time, the timer stop is not so soon. I had to let it run overnight sometimes to simulate the issue.

dubanoze, 2016-07-09:
Guys, hello.
Are you find solution ?
I have same.

dubanoze, 2016-07-11:
Guys, which  version you use ?
i see RTC major update on  1 Jun.
They rewrote rtc & timers code.

MichaelDevid, 2016-08-20:
Any news on this? I have run into issues of the RTC interrupt not firing anymore after a 5-10 seconds of running. But only when I use the STM32L151xE.h include instead of the stm32l151xb.h but my MCU is a stm32l151RE.

adonislixi, 2016-09-18:
We encounter the same problem!
we use "startTime" to light on LED by using STM32l052c8t6 , it can run all night long. 
But, when I chang ping-pong demo ,it only can run first or  several timer. then it never go into RTC interrupt handler........

is that  a bug in SDK??

adonislixi, 2016-09-19:
[https://github.com/Lora-net/LoRaMac-node/issues/123](url)

mluis1, 2016-09-23:
Can we close the issue?

adonislixi, 2016-09-23:
I think so ..

adonislixi@foxmail.com
 
From: Miguel Luis
Date: 2016-09-23 15:49
To: Lora-net/LoRaMac-node
CC: adonis lee; Manual
Subject: Re: [Lora-net/LoRaMac-node] Two or more timers eventually stop (#85)
Can we close the issue?

closed 2016-09-23 by mluis1