Lora-net/LoRaMac-node #88: Sticky Mac commands management can generate invalid frames

ndejean34, 2016-05-31:
Hello,

While testing the "sticky mac commands" feature, I believe I detected an issue concerning the managemetnt of these commands : the device sometimes generates invalid commands by annoucing a given number of piggy-backed bytes and not copying them into the frame.

I will take as reference the commit  "d2c1bbf5d9b7d6b19e1cd90bf6aa01f402bd08f3", labelled "Make MAC commands sticky":
* lines 2739 to 2741 (comment is // Copy the MAC commands which must be re-send into the MAC command buffer): the sticky commands are copied from MacCommandsBufferToRepeat to the end of MacCommandsBuffer and MacCommandsBufferIndex is updated as it should be
* lines 2751 to 2754 : only the first bytes of MacCommandsBuffer are copied into the frame and the additional bytes corresponding to the sticky mac commands are ignored

I propose the following 'simple' correction, implying only line 2751:
* the current code is: `for( i = 0; i < ( MacCommandsBufferIndex - MacCommandsBufferToRepeatIndex ); i++ )`
* and "should" become: `for( i = 0; i < MacCommandsBufferIndex; i++ )`

as MacCommandsBufferIndex is already taking into account the bytes corresponding to the sticky mac commands following its update on line 2741.

Do you agree with my analysis and correction proposal?

Thank you in advance,

djaeckle, 2016-05-31:
Hi ndejean34,

Thanks for your report. I think your analysis is correct. However, I will perform some tests to double check it. I will come back to you when I have more information.

djaeckle, 2016-06-01:
@ndejean34 
I was able to reproduce the issue. I have pushed a fix to the develop branch.

closed 2016-06-03 by mluis1