Lora-net/LoRaMac-node #89: Merging is messy

ChrisEAlfred, 2016-06-02:
There are a lot of changes in master that are very sloppy.

Looking at:
Commit: 63bab8e7470d6685a6c24145f7d5c9175a678ee0 [63bab8e]
Parents: c47feaefd0
Author: Daniel Jäckle <daniel.jaeckle@stackforce.de>
Date: Thursday, 2 June 2016 1:00:48 AM
Issue(#84): Fix issue of CalibrateTimer function.

The file src/boards/SensorNode/board.c only has a few changes, but the whole file was changed. This is not very good practice as it is difficult to determine what was actually the fix/change.

Whenever there is a change, there should only be the smallest possible diff.

djaeckle, 2016-06-02:
Hi ChrisEAlfred,

thanks for the report. I see your point. There are many persons working on the implementation. For the future, when I collect all changes to perform an update of the repository, I will reduce the commits to the smallest possible diffs.

mluis1, 2016-06-02:
We further analyzed the issue and we think that it is due to the usage of different line endings between versions.
If for example you use a tool like WinMerge and set it up to ignore the line endings then you will see only what really changed.
In the future we will pay more attention to this.

Sorry for the inconvenience.

yanfli, 2016-06-03:
This issue is more related to GIT line ending settings,  because of the settings when checkout and push git may covert the line endings for you(UNIX or WINDOWS) not as what you expect, it depends on the real settings.
 And further more if you are using Keil, it will make it more difficult, in such case if your file is using Unix line ending (\n), once you edit it in Keil and save it, then the Keil editor will covert line endings to Windows/DOS one (\r\n) automatically without any caution, very bad, even Keil editor can't recognize UNIX line ending when do whole project searching. 
Because of this I suggest to only use Windows line endings in LoRaMac-Node repo, as most users may prefer to use Keil and develop under Windows.

closed 2016-09-09 by mluis1