Lora-net/LoRaMac-node #90: failing join when setting the node to private

Crossfire60, 2016-06-03:
I'm using a Kerlink SPN base-station, I'm using a private LoRa network and am trying to connect my SK IM880A.
When the network mode is set to public the connection is successful, but when I set it to private, the node will not join the network. I am using the OTA activation method.

pdemil, 2016-06-06:
Try to disable the Public Network on your SK IM880A.
Search  SetPublicNetwork

Crossfire60, 2016-06-07:
Maybe I wasn't clear but that's what I tried which resulted in a failed join. That's why I'm aksing what might have gone wrong.

pdemil, 2016-06-08:
Can you change the configuration of the Kerlink SPN basestation to private network?

mluis1, 2016-09-09:
Can we close the issue?
It looks like that this issue is related to the network setup rather than end-device implementation.

closed 2016-09-23 by mluis1