Lora-net/LoRaMac-node #91: About RTC_ALARM_TICK_DURATION

yanfli, 2016-06-04:
@djaeckle 
 I see in develop branch RTC_ALARM_TICK_DURATION is changed several times. 

Before 5527378d8e1d889b8bec18f4b97805f1704f5b60:
keep 122.07us

5527378d8e1d889b8bec18f4b97805f1704f5b60:
0.48828125 // 1 tick every 488us

4e5a89040384cfc1371d17262d83fc113d48241d
0.48828125 // 1 tick every 488us

2fb46c2f27a35fbb6f5c62bc47d55b436fff04d2
 0.12207031 // 1 tick every 122us

49aa0f9ab26b77a1d6c858dc8f53f4e8dc6905ec
0.48828125      // 1 tick every 488us

From my view, to use slower clock will help to save more power. As ST RTC alarm only supports DATE/HOUR/MINUTE/SECONDS, so the maximum wakeup time is about 1 month, it is set to 28days in the driver, it makes sense because February has 28 days in non leap year. 
This means STM32 MCU will weakup every 28*24*60*60*RTC_ALARM_TICK_DURATION.
If RTC_ALARM_TICK_DURATION  is 0.48828125, it is about 1181s.
If 0.12207031, then 295s. This case takes obviously more power than 0.48828125.

So the question is how to choose RTC_CLK? And how slow this clock could be to keep real time performance (for LoRaWAN stack). Thank you.

daniwebCH, 2016-06-04:
Is your CPU equipped with an RTC that provides the sub second ?
If yes you may consider to not accelerate the RTC tick but use the sub 
second feature...

yanfli, 2016-06-05:
@daniwebCH Thank you for your suggestion! My MCU supports subsecond, it is really helpful, perfect solution.

closed 2016-06-05 by yanfli

daniwebCH, 2016-06-07:
@yanfli : Can you confirm that it worked? Currently I have to much work.. could you provide me the modified files to make working the timer without artificial acceleration but using the subsecond ? my email daniele.wyss@daniweb.ch
Thanks

yanfli, 2016-06-07:
Hi Daniele,
I am sure your proposal solution works. But haven't tried it yet. Too much work as well.

Spectre208, 2016-07-20:
I would like to add that I can confirm that I had the sub-seconds working as you suggested in LoRaMac-node version 4.1.0. 
However, I would have to redo this modification as we have updated to 4.3.0, where the HAL drivers are used instead.

Spectre208, 2016-07-20:
I have dug up an old commit and stripped al other modifications that were made to the file. Please let me know if this works. Also please note that this is made for version 4.0.0.
[rtc-board.h.txt](https://github.com/Lora-net/LoRaMac-node/files/373742/rtc-board.h.txt)
[rtc-board.c.txt](https://github.com/Lora-net/LoRaMac-node/files/373743/rtc-board.c.txt)
(for some reason, i was not able to upload a .zip file, so I added .txt as the file extension)