Lora-net/LoRaMac-node #95: confirmed_message_data_rate_should_decrease_only_when_adr_is_on

yanfli, 2016-06-24:
``` patch
diff --git a/src/mac/LoRaMac.c b/src/mac/LoRaMac.c
index fc8220b..69c09aa 100644
--- a/src/mac/LoRaMac.c
+++ b/src/mac/LoRaMac.c
@@ -1471,7 +1471,7 @@ static void OnMacStateCheckTimerEvent( void )
             {
                 AckTimeoutRetriesCounter++;
 
-                if( ( AckTimeoutRetriesCounter % 2 ) == 1 )
+                if( ( AdrCtrlOn == true ) && ( ( AckTimeoutRetriesCounter % 2 ) == 1 ) )
                 {
                     LoRaMacParams.ChannelsDatarate = MAX( LoRaMacParams.ChannelsDatarate - 1, LORAMAC_TX_MIN_DATARATE );
                 }

```

djaeckle, 2016-06-30:
Hi yanfli,

thanks for your report. We think the implementation is correct as it is. Please refer to chapter 19.4 of the LoRaWAN Specification v1.0.1 - it is not mentioned, that ADR must be enabled for this feature.

yanfli, 2016-06-30:
Thank you for reply. For my understanding, data rate should never change if ADR is off. 
If data rate could change even when ADR is off, then the "ADR off" seems to be useless.

The title of 19.4 is Data Rate Adaption, I think this is a premise of ADR on. How do you think abou this. Because Dare Rate Adaption should only happen when ADR is on.

It will be nice if the author of spec could clarify. @mluis1

kjendal, 2016-07-01:
ADR off means that the network server should not expect that the end device will accept ADR commands (the server also then expects that the end-device does not autonomously change its own data rate, but this isn't a failure condition, rather it leads to a lack of optimization).  It does not mean that the end-device cannot change its data rate based on its local knowledge of conditions or application requirements.  As such, the code as written is correct.

yanfli, 2016-07-01:
@kjendal Thank you for your answer.

mluis1, 2016-07-01:
Thanks @kjendal for giving a more precise answer. I'm sorry I didn't answer earlier.

yanfli, 2016-07-01:
Thank you, all. As you all agree, I'll close this issue.

closed 2016-07-01 by yanfli