Lora-net/LoRaMac-node #96: Uart TX problem

dubanoze, 2016-06-28:
Guys, hello.
Please check Uart TX. Started work only after this monkey patch.

bool TxReady = false;
uint8_t UartMcuPutChar( Uart_t *obj, uint8_t data )
{
    if( IsFifoFull( &obj->FifoTx ) == false )
    {
	//__HAL_UART_DISABLE_IT( &UartHandle, USART_IT_TXE );
        __disable_irq( );
        FifoPush( &obj->FifoTx, data );
        __enable_irq( );
        // Enable the USART Transmit interrupt
				if (TxReady == false) {
						TxReady = true;
						HAL_UART_Transmit_IT(&UartHandle, &data, 1 );
				}
        //__HAL_UART_ENABLE_IT( &UartHandle, USART_IT_TXE );
        return 0; // OK
    }
    return 1; // Busy
}

Regimbal, 2016-07-13:
Same problem here, UART does not work. Applied the patch suggested but did not do the job in my case. Will look further.
I suspect while upgrading to HAL, peripherals initialization through HAL have not been done properly. Already observed it with the ADC, mentioned in an other Issue post.

dubanoze, 2016-07-13:
I sent one part patch. 
Now i rewrite uart-board to DMA & timers.

dubanoze, 2016-07-13:
void HAL_UART_TxCpltCallback( UART_HandleTypeDef *UartHandle )
{
    uint8_t data;

    if( IsFifoEmpty( &Uart1.FifoTx ) == false )
    {
				data = FifoPop( &Uart1.FifoTx );
				HAL_UART_Transmit_DMA(UartHandle, &data, 1 );
    }
    else
    {

        // Disable the USART Transmit interrupt
        //HAL_NVIC_DisableIRQ( USART1_IRQn );  ////
    }
    if( Uart1.IrqNotify != NULL )
    {
        Uart1.IrqNotify( UART_NOTIFY_TX );
    }
}

x893, 2016-07-13:
very none efficient code

Spectre208, 2016-07-19:
There is still an issue where the first character to be put in an empty fifo gets transmitted twice, because it both gets passed to the HAL and pushed onto the fifo. I have fixed it with a slightly modified version of your patch below:

```c
bool TxBusy = false;
```

```c
uint8_t UartMcuPutChar( Uart_t *obj, uint8_t data )
{
    if( IsFifoFull( &obj->FifoTx ) == false )
    {
        if ( TxBusy == true ) {
        	__disable_irq( );
			FifoPush( &obj->FifoTx, data );
			__enable_irq( );
			
		} else
		{
			TxBusy = true;
			// Enable the USART Transmit interrupt
			HAL_UART_Transmit_IT(&UartHandle, &data, 1 );
		}
        return 0; // OK
    }
    return 1; // Busy
}
```

```c
void HAL_UART_TxCpltCallback( UART_HandleTypeDef *UartHandle )
{
    uint8_t data;

    if( IsFifoEmpty( &Uart1.FifoTx ) == false )
    {
        data = FifoPop( &Uart1.FifoTx );
        //  Write one byte to the transmit data register
        HAL_UART_Transmit_IT( UartHandle, &data, 1 );
    }
    else
    {
    	TxBusy = false;
    }
    if( Uart1.IrqNotify != NULL )
    {
        Uart1.IrqNotify( UART_NOTIFY_TX );
    }
}
```

dubanoze, 2016-07-20:
I rewrite uart-board.c with DMA.
Now support 115 200 without ORE and lost symbols.
I use Uart with Quectel L76. 

```C
#include "board.h"

#include "uart-board.h"

//
UART_HandleTypeDef UartHandle;
//
DMA_HandleTypeDef hdma_usart1_rx;
DMA_HandleTypeDef hdma_usart1_tx;
//
tUartData UartData;

void RxTimeoutIrq ( void );

void UartMcuInit( Uart_t *obj, uint8_t uartId, PinNames tx, PinNames rx )
{
	
    obj->UartId = uartId;

    __HAL_RCC_USART1_FORCE_RESET( );
    __HAL_RCC_USART1_RELEASE_RESET( );

    __HAL_RCC_USART1_CLK_ENABLE( );

    GpioInit( &obj->Tx, tx, PIN_ALTERNATE_FCT, PIN_PUSH_PULL, PIN_PULL_UP, GPIO_AF7_USART1 );
    GpioInit( &obj->Rx, rx, PIN_ALTERNATE_FCT, PIN_PUSH_PULL, PIN_PULL_UP, GPIO_AF7_USART1 );
	
}

void UartMcuConfig( Uart_t *obj, UartMode_t mode, uint32_t baudrate, WordLength_t wordLength, StopBits_t stopBits, Parity_t parity, FlowCtrl_t flowCtrl )
{
	
    UartHandle.Instance = USART1;
    UartHandle.Init.BaudRate = baudrate;

    if( mode == TX_ONLY )
    {
        if( obj->FifoTx.Data == NULL )
        {
            assert_param( FAIL );
        }
        UartHandle.Init.Mode = UART_MODE_TX;
    }
    else if( mode == RX_ONLY )
    {
        if( obj->FifoRx.Data == NULL )
        {
            assert_param( FAIL );
        }
        UartHandle.Init.Mode = UART_MODE_RX;
    }
    else if( mode == RX_TX )
    {
        if( ( obj->FifoTx.Data == NULL ) || ( obj->FifoRx.Data == NULL ) )
        {
            assert_param( FAIL );
        }
        UartHandle.Init.Mode = UART_MODE_TX_RX;
    }
    else
    {
       assert_param( FAIL );
    }

    if( wordLength == UART_8_BIT )
[... 251 lines cut ...]

void UartRxDisable ( void ) {
			
		HAL_UART_DMAStop(&UartHandle);
		__HAL_UART_DISABLE_IT(&UartHandle, UART_IT_ERR);
		HAL_NVIC_DisableIRQ( USART1_IRQn );
	
}

void RxTimeoutIrq ( void ) {
	
		UartData.RxCount++;
		HAL_UART_DMAStop(&UartHandle);
		
		if( Uart1.IrqNotify != NULL )
		{
			Uart1.IrqNotify( UART_NOTIFY_RX );
		}		
		memset(UartData.RxData, 0, sizeof(UartData.RxData));
		__HAL_UART_FLUSH_DRREGISTER(&UartHandle);				
		if (HAL_UART_Receive_DMA(&UartHandle, UartData.RxData, 512 ) != HAL_OK) {  }				

}

```

mluis1, 2016-09-09:
Can we close the issue?

dubanoze, 2016-09-09:
I think yes.

closed 2016-09-09 by mluis1