Lora-net/LoRaMac-node #98: LoRaMAC data store to EEPROM

shankuuuu, 2016-06-29:
When the server asks the modem to add/remove a channel or change RX2 frequency/datarate or set delay between Tx and Rx, do we need to write these values into a persistent memory? Currently in LoRaMac.c I don't see any of this values getting stored. So after a modem reset these values will be lost. Is it OK to have this behavior or do we need to add mechanisms to store these changes somewhere? 

Is it mandatory by LoRaWAN standards to enable this features? Can we NACK the server for these requests?

If we have to store the details in EEPROM, what is the list of items that we are expected to store so as to be compatible with LoRaWAN standards?

kjendal, 2016-06-29:
The RX values that the network server configures on the end-device MUST be persisted as long as the device remains joined to the server (i.e. is using the same devaddr, session keys and FCnt sequence).  If the device rejoins it must revert the RX parameters to their default values.  The same SHOULD be done for the TX parameters (datarate and channel mask), but the device can recover from the loss of this information.

The device may NACK some of these commands, however, the network server is commanding the device to these values for the devices own good (to improve downlink success).

Dave Kjendal
Senet, Inc.

shankuuuu, 2016-06-30:
@kjendal  thanks for the advice. Currently I am not storing Up-link Counter value when end device powers off. I am using Activation By Personalisation. So the server should consider it as a reset when it receives an uplink with FCnt seq as 1. Am I right? But I dont see the server resetting the Down-link counter. Does that mean we have to store the down-link counter value every time in the persistent memory? Any help on this will be greatly appreciated.

kjendal, 2016-07-01:
You are required to not reuse FCnts, so you need to be storing that. To do otherwise increases the probability that your keys can be compromised through observation.

mluis1, 2016-09-23:
We close this one as it is a duplicate of #59 issue.

closed 2016-09-23 by mluis1