Lora-net/LoRaMac-node #100: RTC timing accuracy

mpt8153, 2016-06-30:
I've a question about RTC timing accuracy. In this implementation RTC has a tick every 488 microseconds. In LoRaWan Specification is written that RX1 and RX2 windows are open at RECEIVE_DELAY1 and RECEIVE_DELAY2 (+/- 20 microseconds). So with this implementation is not possible to guarantee to open an RX window with +/- 20 microseconds of tollerance. Is this right?

kjendal, 2016-07-01:
You should expect that the downlink is actually present with +/- 20usec of the actual correct time.  If your RTC is not very accurate, then you need to open up the receiver on/off times to make sure you are listening during the correct time.  This will have a minor impact on your power consumption.

mpt8153, 2016-07-01:
@kjendal Thank you.

closed 2016-07-01 by mpt8153