Lora-net/LoRaMac-node #103: TimerStart(), TimerStop() functions unconditionally enable interrupts on return

catsup, 2016-07-01:
The TimerStart() and TimerStop() functions (in src/system/timer.c) disable interrupts upon entry, and re-enable interrupts when returning, regardless of whether interrupts were disabled or not before entry.

This can cause problems if the calling flow disabled interrupts and relies on them to remain disabled.

Interrupts should only be re-enabled if they were previously disabled (ie by reference counting or checking interrupt status upon function entry).

pdemil, 2016-07-16:
> Interrupts should only be re-enabled if they were previously disabled

Interrupts should only be re-enabled if they were previously enabled

catsup, 2016-07-16:
Yes, obviously. Thanks for the correction.

mluis1, 2016-09-09:
We will try to provide a fix for this as soon as possible.

closed 2016-09-30 by mluis1