Lora-net/LoRaMac-node #105: high stop mode consumption with im880

Regimbal, 2016-07-05:
Dear all,
I developed a board using then im880 and based my app on the the SK-im880a loramac class A sources.
I made the required changes in board.c and board.h to match with my pinout. Also made the necessary modifications in main.c so that I do not get any compilation error and that's it.
With this configuration, I observe a STOP mode current around 140uA instead of something closer to 2uA.
Cannot find why... Any idea?

Thanks a lot

dubanoze, 2016-07-05:
Hi, i can confirm about 2uA in STOP mode on   iM880(A)-L.

Regimbal, 2016-07-05:
This is good to know...

jmalangoni, 2016-07-05:
did you define LOW_POWER_ENABLED?

Regimbal, 2016-07-06:
Hi everyone,
- So I checked my GPIOs and appended unused gpio as analog input: no difference.
- Second, i had already undefined USE_DEBUGGER but hadn't seen any difference. I do now. Actually I used to flash the uC with STlink software with 'low power mode debugger' option checked. Unchecking the corresponding box made my consumption fall to ~40uA.
- Third, I have not defined LOW_POWER_ENABLED. Haven't seen any reference to it in the last release of the sources (4.3.0). But I might be mistaken. In the 4.0 release we had LOW_POWER_MODE_ENABLED defined... In 4.3 (src for board SK-im880A), it has been replaced by a definition of the power supply source (battery).

Thanks for the help

Spectre208, 2016-07-20:
Are you still having a high stop mode power consumption? 
40uA sound like a lot, anyway.
Are you by chance using SPI and/or the GPIO P15 pin on the iM880A?

Regimbal, 2016-07-20:
On this board, yes I still get the 40uA.
I'm not using SPI, but P15 yes! It is defined as
`GpioInit( &Alm1, ALM_1, PIN_INPUT, PIN_PUSH_PULL, PIN_NO_PULL, 1 );`
and affected to an interrupt to wake the uC module from low power stop.
I think I have another board which does not use P15. But never tested the current consumption on this one...
Could you tell me why P15 would lead to extra power consumption?

Thanks for the help

Spectre208, 2016-07-20:
Due to a bug in De-initializing SPI (which happens because the SX1272 chip is connected using SPI), the mcu pin name with value 0 gets set as a high output (as reported in issue #111). This value happens to reference to pin P15 on the iM880A module. When entering stop mode, it feeds current to whatever is connected to that pin.

You can try to fix this by adding a pin assignment in spi-board.c like this:
```c
if( nss != NC )
{
	GpioInit( &obj->Nss, nss, PIN_ALTERNATE_FCT, PIN_PUSH_PULL, PIN_PULL_UP, GPIO_AF5_SPI1 );
}
else
{
	obj->Spi.Init.NSS = SPI_NSS_SOFT;
	obj->Nss.pin = NC;
}
```

Regimbal, 2016-07-26:
Found my mistake on a Gpio which had escaped my attention.
Now I get the expecte consumption.
No issue with PA0 though...

closed 2016-07-26 by Regimbal