Lora-net/LoRaMac-node #109: ADREnabled

reissjason, 2016-07-12:
Should LinkADRReq be ignored if ADR is not enabled?

If not how does the network server know the power level of the device in order to send the correct command?

With ADR and ACK disabled and after a LinkADRReq to change the channel mask if a device is left unable to connect to the network there is not a path to re-enable the default channels.

kjendal, 2016-07-13:
1) In general, the network server will not send LinkADRReq to the device if it has signaled that ADR is disabled.  By clearing the ADR bit in the uplink the device is signalling to the network server that it intends to manage its own DR, TXPower and Channel Plan and is likely to ignore any LinkADRReq received.

2) In LoRaWAN 1.0.x, the network only knows the default TX Power via out-of-band configuration.  If it is not managing the device through ADR, then the TX Power is not relevant to the operation of the network server.

3) It is implied that the device is managing its own channel plan if the ADR bit is disabled.  The device may use whatever means available to it to determine that channel plan, and that may require out-of-band communication between the solution provider and the network.

reissjason, 2016-07-13:
OK, show channel plan administration through a ADR commands is not
independent of ADR datarate usage.

Outright rejection of ADR commands when ADR is disabled would not be
incorrect behavior although it shouldn't be needed since the device should
not receive the commands in the first place.

closed 2016-07-13 by reissjason