Lora-net/LoRaMac-node #110: UartPutBuffer Error Handling

Spectre208, 2016-07-20:
The function UartPutBuffer in uart.c does not take the current baud rate into account when putting a buffer to UART. It simply tries to put a character into the buffer 10 times before aborting the operation. 
```c
/*!
 * Number of times the UartPutBuffer will try to send the buffer before
 * returning ERROR
 */
#define TX_BUFFER_RETRY_COUNT                       10
```

```c
while( UartPutChar( obj, buffer[i] ) != 0 )
{
	retryCount++;

	// Exit if something goes terribly wrong
	if( retryCount > TX_BUFFER_RETRY_COUNT )
	{
		return 1; // Error
	}
}
```
However, the function UartPutChar does not return an error, it returns whether it is busy or not. When the baudrate is set to 115200 and the transmit buffer is full, it takes longer than 10 tries before a spot in the buffer becomes available. To me, this is not when "something goes terribly wrong", because it is just required to wait a little longer.

When UartPutBuffer aborts the operation, there is no way of telling where it stopped trying. If it returned the amount of characters it has already put into the buffer, one could adjust the buffer pointer and try again later. 

Maybe the TX_BUFFER_RETRY_COUNT setting should automatically change, depending on how long it takes to send a single character through UART, thus depending on the current baudrate. Increasing this setting improved the stability of the UartPutBuffer function for me.

Regimbal, 2016-07-21:
How much tries would you set for a bitrate of 115200 then?

9910292559, 2016-07-26:
use some   delay
its work fine

closed 2016-09-23 by mluis1

Spectre208, 2016-09-23:
Has this been fixed?

The function would be way more reliable if it returned the character count it successfully put into the buffer

mluis1, 2016-09-23:
No, we haven't fixed it because we think that there is no need to take the baudrate in account for this function. In plus there is a FIFO in place which should handle the transmission delay by itself.

The only advice we can provide is maybe to increase the FIFO size or the number of retries in case your MCU runs very fast.

Spectre208, 2016-09-23:
Taking the baudrate into account for this function was just one of my suggestions for the issue, where the actual issue is the incorrect error handling.

We need to keep the FIFO size low to save space for other functionality. Increasing the number of retries seems more of a workaround to the problem than it is the solution.

Thanks for your response, anyway.