Lora-net/LoRaMac-node #115: RTC priority change

FoujanetG, 2016-07-25:
Hi,

I see that the RTC alarm priority( preemption bit) has been changed with the new release.
In, rtc-board.c , RTCInit() :
In 4.1, it is `HAL_NVIC_SetPriority( RTC_Alarm_IRQn, 0, 0 );,`
In 4.2 it is `HAL_NVIC_SetPriority( RTC_Alarm_IRQn, 4, 0 );`

What is the reason of this change?

Thanks, 
GF

pdemil, 2016-11-15:
and in the example zip here ( https://github.com/Lora-net/LoRaMac-node/issues/123#issuecomment-248950792 ) it is changed to:
 HAL_NVIC_SetPriority( RTC_IRQn, 1, 0 ); 
(note another name is used for the IRQn).

GregCris, 2017-01-25:
Setting the IRQ priority for a complete system is quite challenging and in the end, it is down to each user to review the priorities depending on the various source of IRQs. If you review the complete IRQ priorities for the LoRaMote for example, the priority 4 is the second highest one after only the systick and all the other IRQ will have a lower priority.

mluis1, 2017-02-03:
Can we close this issue?