Lora-net/LoRaMac-node #118: Problem with datarate values for tx messages

mpt8153, 2016-07-29:
I think that in  LoRaMacQueryTxPossible this line:
int8_t datarate = LoRaMacParamsDefaults.ChannelsDatarate;
should be changed with
int8_t datarate = LoRaMacParams.ChannelsDatarate;

@ 19.4 Data-Rate Adaptation during Message Retransmissions
"For example if an end-device sends a “confirmed” frame first using DR5 and has to
retransmit 3 times (twice at DR5 then twice at DR4), the next frame transmitted will use DR4"

Is this correct?

BR

mpt8153, 2016-08-23:
Any news about that?
Perhaps I didn't explay well what hapen.
In LoRaWan Specification at chapter 19.4 is written: " Any further transmission uses the last data rate used. For example if an end-device sends a “confirmed” frame first using DR5 and has to retransmit 3 times (twice at DR5 then twice at DR4), the next frame transmitted will use DR4".
In LoRaWan stack version 4.3 I've found that next frame will be transmitted with default datarate and not with the datarate derived from previous retransmission frame. Taking the LoraWAN specification example with LoRaWan stack version 4.3 the next frame will be transmitted with DR5 and not DR4.
Is this correct?

djaeckle, 2016-08-24:
Hi mpt8153,

thanks for the report. I think I was able to reproduce when I disable ADR. When the node sends the next frame after retransmissions of the previous one, the main application sets `LORAWAN_DEFAULT_DATARATE` as datarate in function `SendFrame`. Is this your use case?

mpt8153, 2016-08-24:
Thanks for you reply. Yes it is my case. As you have suggested it may be solved changing main application. 
But I have another question. With ADR disabled if application will use last DR for a new transmission, in application life the DR continues to decrease due to the possibility of multiple retry. Following indication of 19.4 chapter of LoRaWan specification there is no way to increase DR. I think it is a lack of LoRaWAN specification. What do you think?

djaeckle, 2016-08-24:
Hi mpt8153,

we will discuss this issue and see how we solve it. However, it could take some time due to the summer break.
I think your statement is correct. The specification is not clear about it. From my point of view, if the node has received a downlink frame from the server, the node could possibly apply a procedure to increase the datarate. But I think the implementation is up to the application.

mpt8153, 2016-08-24:
Thank you very much for your immediate answer.

djaeckle, 2016-09-20:
Hi mpt8153,

we discussed the issue and we agreed that the MAC layer itself should not be responsible to change the datarate in this case. Thus, it is up to the application.

closed 2016-09-23 by mluis1