Lora-net/LoRaMac-node #120: Where I can buy Cherry LCC Nymtek module?

manhviking, 2016-08-06:


manhviking, 2016-08-06:
I have seen the Sensor Node Schematic, it used Cherry LCC Nymtek module. But I searched google don't have good result.
Schematic of module and iM880A is really different
Please help me
Thank

mluis1, 2016-08-10:
Please find below the answer that I got from my colleagues in China.

The Cherry module is made by Nymtek (a company in Nanjing China) but which is obsolete and not available any more.
So, in the latest version of the SensorNode hardware (V0.4) we do not use the module any more but put SX1276 part directly on board.

We will update the SensorNode schematics for the next release. In the mean time I have attached the new schematic here.

[SensorNode-schematic.zip (V0.4)](https://github.com/Lora-net/LoRaMac-node/files/410647/SensorNode-schematic.zip)

manhviking, 2016-08-10:
Thank you very much friend

manhviking, 2016-08-10:
Do you know any Lora-gateway hardware that have schematic or something like that, that can work with this sensor friend?

manhviking, 2016-08-10:
I can not download that file
Please send me to my gmail: manhnv.etk56@gmail.com
Thank you friend

closed 2016-09-23 by mluis1