Lora-net/LoRaMac-node #121: Frame pending flag on port 0 handling, stack or application handling?

e123h, 2016-08-09:
Dear all, 

If an gateway sends MAC downlink message on fport 0 with a pending frame flag = 1. Do I need to handle this within my own application? Normally all fport 0 messages are handled within the stack (channel updates etc.) , but I couldn't find the frame pending flag handling?

thanks in advance.

mluis1, 2016-08-09:
The Fpending flag must be handled at the application layer. This flag just indicates that the server has more data to send to the end-device. 
It is up to the application to decide if it is required to create new possibilities for a downlink to take place by sending uplinks as soon as possible.

The Fpending information is always sent to the application layer regardless of the used port.

This flag can be handled as indicated in chapter 18.3 or 19.3 (depends on revision of the document) of the LoRaWAN specification:
>  If a frame with the FPending bit set requires an acknowledgement, the end-device shall do so as described before. If no acknowledgment is required, the end-device may send an empty data message to open additional receive windows at its own discretion, or wait until it has some data to transmit itself and open receive windows as usual.

e123h, 2016-08-09:
mluis1, 

Thank you for your quick response.

e123h, 2016-08-09:
Is this is also true for ACK flag on confirmed data down messages?  

> ....The network uses the downlink receive window to
> transmit a ―'confirmed' data frame towards the end-device on the same channel A. Upon
>  reception of this data frame requiring an acknowledgement, the end-device transmits a
>  frame with the ACK bit set at its own discretion....  
(LoRaWAN spec. 1.0)

Because the situation occurs that if I periodically send messages and not acknowledge an confirmed down message  I get stuck within while trying to transmit uplink messages;

LoRaMacStatus_t LoRaMacMcpsRequest( McpsReq_t *mcpsRequest )
{
    LoRaMacStatus_t status = LORAMAC_STATUS_SERVICE_UNKNOWN;
    LoRaMacHeader_t macHdr;
    uint8_t fPort = 0;
    void *fBuffer;
    uint16_t fBufferSize;
    int8_t datarate;
    bool readyToSend = false;

    if( mcpsRequest == NULL )
    {
        return LORAMAC_STATUS_PARAMETER_INVALID;
    }
    **if( ( ( LoRaMacState & MAC_TX_RUNNING ) == MAC_TX_RUNNING ) ||
        ( ( LoRaMacState & MAC_TX_DELAYED ) == MAC_TX_DELAYED ) )
    {
        return LORAMAC_STATUS_BUSY;
    }**

with ' MAC_TX_RUNNING' being '1'.  
The only place this status flag will be reset is within the function 'OnMacStateCheckTimerEvent' and none of the arguments are valid and will result in a endless loop.

e123h, 2016-08-09:
I'm sorry this also occurs when I do acknowledge the confirmed down messages. Its to reproduce because its not always the case. But its still strange that it falls within that endless loop and non of the arguments (if-statements) within the 'OnMacStateCheckTimerEvent' are valid to act on the 'MAC_TX_RUNNING' flag.

closed 2016-08-19 by e123h