Lora-net/LoRaMac-node #122: Ignoring incorrect frames

e123h, 2016-08-15:
Dear all, 

We are currently testing a LoRaWAN node with 4.3 version of stack, on a nation wide network. 
The LoRaWAN 4.3 stack 'hangs' on the following situation; The network sends faulty 'MAC confirmed downlink messages' where the FOptsLen is different from 0 on FPort 0.  (The Network operator is informed about the situation). The problem is that the current stack version 4.3 is not compliant with the latest addition of the specification 'Should this occur, the device shall ignore the frame.'  

LoRaWAN specification 1.0 / 1.0.1: 
> 6 If FOptsLen is 0, the FOpts field is absent. If FOptsLen is different from 0, i.e. if MAC
> 7 commands are present in the FOpts field, the port 0 cannot be used (FPort must be either
> 8 not present or different from 0).
> 9 MAC commands cannot be simultaneously present in the payload field and the frame
> 10 options field.

In addition to 1.0 within 1.0.1:
> Should this occur, the device shall ignore the frame.

After entering the IF-statement on line 1228 in LoRaMac.c the stack continues to the ELSE statement (because FOptsLen is different from 0) where 'skipIndication = true;'  after this the stack hangs with 'MAC_TX_RUNNING'.

e123h, 2016-08-19:
for example, the 'payload' variable (on line 1223) contains the following data: 

0xa0 0x0 0xe 0x0 0x14 0x36 0x3 0x0 0x7 0x7 0x58 0x6e 0x84 0x50 0x0 0xc6 0xd2 0x25 0x19 0x9e 0xb0 0x17 0x57 0x21 0xcd 0xda 0xf 0x30 0xbd 0x2f 0x65 0xc 0x74 0xec 0x31 0x65 0x0   

This is captured on line 1223, this is after the MIC check. 

After receiving this packet, any attempt to send a new packet fails. If I don't set the flag 'skipIndication' on line 1245 the stack is still able to sent packages. But in this situation the stack will still acknowledge the packet which is an error.

djaeckle, 2016-08-24:
Hi e123h,

thanks for the report. I was able to reproduce the issue. The issue occurs when the node transmits confirmed uplinks and receives a frame on port 0 with fOpts > 0. I think the problem is that in the function `OnRadioRxDone`, the MAC stops the timer `OnAckTimeoutTimer`. However, the MAC should not stop it when it ignores a frame. We will provide an update for this.

closed 2016-09-30 by mluis1