Lora-net/LoRaMac-node #123: RTC Interrupts stops after 5-10 seconds

MichaelDevid, 2016-08-20:
Hi All,

For our hardware I have changed the MCU to a STM32L151xE type instead of the STM32L151xB. Now when upload the firmware without any changes it runs fine. 

But when I change the build define of the MCU type to STM32L151xE (I have added the header file stm32l151xe.h version V2.1.2 same version as the stm32l151xb.h) the RTC interrupt for handling all the timer events stops firing after 5-10 seconds of running. 

I'm using the latest master commit version 4.3.0

Any ideas of why this is happening? 

Thanks,

Michael

MichaelDevid, 2016-08-20:
I have found the issue. The RtcStartWakeUpAlarm should also set the SubSecond values for certain MCU types. I suggest the RtcStartWakeUpAlarm is changed to:

`static void RtcStartWakeUpAlarm( uint32_t timeoutValue )
{
    RtcCalendar_t now;
    RtcCalendar_t alarmTimer;
    RTC_AlarmTypeDef alarmStructure;

    HAL_RTC_DeactivateAlarm( &RtcHandle, RTC_ALARM_A );
    HAL_RTCEx_DeactivateWakeUpTimer( &RtcHandle );

    // Load the RTC calendar
    now = RtcGetCalendar( );

    // Save the calendar into RtcCalendarContext to be able to calculate the elapsed time
    RtcCalendarContext = now;

    // timeoutValue is in ms 
    alarmTimer = RtcComputeTimerTimeToAlarmTick( timeoutValue, now );

    alarmStructure.Alarm = RTC_ALARM_A;
    alarmStructure.AlarmDateWeekDaySel = RTC_ALARMDATEWEEKDAYSEL_DATE;
    alarmStructure.AlarmMask = RTC_ALARMMASK_NONE;
    alarmStructure.AlarmTime.TimeFormat = RTC_HOURFORMAT12_AM;
    
	#if defined(STM32L100xBA) || defined (STM32L151xBA) || defined (STM32L152xBA) || defined(STM32L100xC) || defined (STM32L151xC) || defined (STM32L152xC) || defined (STM32L162xC) || defined(STM32L151xCA) || defined (STM32L151xD) || defined (STM32L152xCA) || defined (STM32L152xD) || defined (STM32L162xCA) || defined (STM32L162xD) || defined(STM32L151xE) || defined(STM32L151xDX) || defined (STM32L152xE) || defined (STM32L152xDX) || defined (STM32L162xE) || defined (STM32L162xDX)
    alarmStructure.AlarmSubSecondMask = RTC_ALARMSUBSECONDMASK_ALL;
    alarmStructure.AlarmTime.SubSeconds = 0;
	#endif

    alarmStructure.AlarmTime.Seconds = alarmTimer.CalendarTime.Seconds;
    alarmStructure.AlarmTime.Minutes = alarmTimer.CalendarTime.Minutes;
    alarmStructure.AlarmTime.Hours = alarmTimer.CalendarTime.Hours;
    alarmStructure.AlarmDateWeekDay = alarmTimer.CalendarDate.Date;

    if( HAL_RTC_SetAlarm_IT( &RtcHandle, &alarmStructure, RTC_FORMAT_BIN ) != HAL_OK )
    {
        assert_param( FAIL );
    }
}
`

MichaelDevid, 2016-09-10:
Why did you close this? I think you should really add this because it is part of the HAL.

belovictor, 2016-09-10:
Your fix also worked for us on STM32L151CCT6. Without it we observe the same problem - RTC interrupts stop working after a random period of time.

mluis1, 2016-09-12:
Your fix for the SubSeconds looks valid even though it may also be necessary to fully handle the SubSeconds field in order to not loose time precision.

For your information the boards directory files are meant for a specific end-device hardware thus a specific MCU. None of the current used MCU has the SubSecond that's why your fix isn't included.

If I'm not wrong the latest version of the used MCU have the SubSecond thus, what we could do is to add the specific conditional pre-processing for this new MCU version.

We are currently developing a new end-device hardware that includes an MCU with the SubSencond field which when ready will provide an example on how to fully handle the SubSecond field.

MichaelDevid, 2016-09-12:
Thanks for your elaboration! 

Could this "not handling the SbuSecond correctly" issue be linked to this other issue I have? 

[RX2 Issue](https://github.com/Lora-net/LoRaMac-node/issues/131)

mluis1, 2016-09-12:
It may be a reason. But, as you said your timing measurements look right.

In order to debug what is going on I would advise you to
1. Verify the second window parameters. (Datarate and Frequency)
2. If the previous are correct you may try to open the windows earlier in time and for longer time.
3. If you get the downlinks working then it may mean that the RTC timings aren't precise enough.

Hope it helps you find a solution for your issue.

MichaelDevid, 2016-09-18:
Is it possible to get some development version of this SubSecond example? Or do you have any indication on when this will be released?

adonislixi, 2016-09-19:
HI,  @mluis1 ,we encounter the same problem.
we use STM32L072/STM32L052.  we found RTC interrupt stop after several times.
and the "subseconde issues " does not work for us...
Is that the sdk RtcStartWakeUpAlarm  can not work on STM32L0xx MCU ?

nestorayuso, 2016-09-19:
STMicroelectronics recently created a fork of LoRaMac-node with support of more STM32 families. 
Maybe this can help.

http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-expansion-software/i-cube-lrwan.html

adonislixi, 2016-09-20:
the STMicroelectronics's code works.. 
Rtc_subseond needs to be set in Stm32L0xx MCU

thank u...

adonislixi@foxmail.com
 
From: Nestor Ayuso
Date: 2016-09-19 17:09
To: Lora-net/LoRaMac-node
CC: adonis lee; Comment
Subject: Re: [Lora-net/LoRaMac-node] RTC Interrupts stops after 5-10 seconds (#123)
STMicroelectronics recently created a fork of LoRaMac-node with support of more STM32 families. 
Maybe this can help.
http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-expansion-software/i-cube-lrwan.html

mluis1, 2016-09-22:
Currently we are working on a new platform that is based on the STM32L051C8 which has the sub

mluis1, 2016-11-29:
We close this issue as we just pushed the MoteII platform support which is based on the STM32L051C8 MCU.

closed 2016-11-29 by mluis1