Lora-net/LoRaMac-node #125: End Node stop sending new packets if duplicated confirmed downlink message arrives

shankuuuu, 2016-08-24:
while testing I noticed that end node hangs and stops sending further messages if duplicated confirmed downlink message arrives with downlink counter != 0.

                        if( macHdr.Bits.MType == FRAME_TYPE_DATA_CONFIRMED_DOWN )
                        {
                            SrvAckRequested = true;
                            McpsIndication.McpsIndication = MCPS_CONFIRMED;

                            if( ( DownLinkCounter == downLinkCounter ) &&
                                ( DownLinkCounter != 0 ) )
                            {
                                // Duplicated confirmed downlink. Skip indication.
                                skipIndication = true;
                            }

It goes to the skipIndication = true, but after that we will not be able to send further uplink messages.

djaeckle, 2016-08-24:
Hi shankuuuu,

thanks for the report. Do you use confirmed or unconfirmed uplinks?

shankuuuu, 2016-08-24:
Hi djaeckle,

I sent confirmed uplinks.

djaeckle, 2016-08-24:
Hi shankuuuu,

thanks. I guess the problem is similar to #122. However, I will double check it.

9910292559, 2016-08-24:
Dear all
i am using WiMOD_LR_Base_iU880A_V1_10 firmware as Base station and sending join Request from end device, but no Ack recieved.
Please some help and  what may be  wrong with it .
I am looking received Ack frame on DIO0 interrupt

djaeckle, 2016-08-24:
Hi 9910292559,

thanks for the report. Please provide some more details about your issue. It seems like it does not fit to this thread. If so, please open up a new issue.

closed 2016-09-30 by mluis1